arduino/Arduino #5215: Mouse continually 'clicked' by Serial Monitor window

frankvdh, 2016-08-06:
Ubuntu 16.04LTS, Arduino 1.6.10

Something (I think opening the Serial Monitor window in the Tools menu) causes the mouse to behave as if it is continually being clicked. It seems also to be moved to the left edge of the screen, outside the Arduino windows, where I have my start menu. So moving in the Serial Monitor window causes dozens of other apps to start, without physically clicking the mouse.This makes the Serial Window unusable.

It seems to be repeatable by opening the Serial Monitor window immediately after starting the Arduino app.

facchinm, 2016-08-10:
Hi @frankvdh ,
I've never seen anything similar to this issue. The only thing that comes to my mind is that you may have a Micro/Leonardo attached with an HID (Mouse + Keyboard) sketch loaded, and the sketch is waiting for the serial monitor to open with the `while !Serial()` line.
Once you open the serial monitor the board starts acting like a mouse and click thing around.
Could you confirm/post a video of the event? Thanks

sandeepmistry, 2016-12-12:
Closing this for now due to lack of feedback.

closed 2016-12-12 by sandeepmistry

cmaglie, 2018-08-13:
Is the Wemos running a sketch that emulates a Mouse?

frankvdh, 2018-08-31:
I just tried it with a brand-new "Genuine" Arduino Uno (probably a Chinese clone), just plugged it in, started the IDE, changed board to Genuino Uno, opened the Blink example, uploaded to the board,  and the same thing occurs. However, it doesn't seem as bad as the ESP example. I can move the mouse around quite a bit with only occasional "clicks" happening.

PaulStoffregen, 2018-09-02:
In a terminal, try running "lsusb".  Look for which line is your board.  With a truly genuine Arduino Uno R3, I see this:

    Bus 001 Device 014: ID 2341:0043 Arduino SA Uno R3 (CDC ACM)

The point is to get the 8 digit ID number, in this case "2341:0043".

Then run "lsusb -v -d 2341:0043" (but with whatever number your board uses).  That will print lots of info about the USB info for your board.  Please post it here, so we can see the properties of the hardware you're really using.

To keep things readable, look for the "<>" button (Insert code if you hover your mouse).  Click this, then paste all that text *after* the characters it adds.  This will preserve the white space, so we can read your USB info easily.  Please use the "Preview" tab to check the result will be readable when you post the comment.

frankvdh, 2018-12-25:
It appears that this is an Ubuntu issue. I see similar symptoms if I plug my ESP8266 into the USB port after restarting Ubuntu and logging in, *before* starting the Arduino IDE. So this issue can be closed.

Also, in case anyone else runs into this, this only occurs if my USB mouse is plugged into the same USB hub as the ESP8266. If either is plugged into a USB port on the PC itself, the problem doesn't happen.