arduino/Arduino #5216: compilation errors on folders with Korean names

Myriads, 2016-08-07:
Compiling the sketch on the Korean named folder, the following error messages are displayed:

> 스케치를 컴파일 중…
> "D:\Tools\arduino-1.6.10\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Administrator\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.12\cores\arduino" "-IC:\Users\Administrator\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.12\variants\standard" "C:\Users\ADMINI~1\AppData\Local\Temp\build23759f5b464ad05d4eee1a2a8893d847.tmp\sketch\Ping.ino.cpp" -o "C:\Users\ADMINI~1\AppData\Local\Temp\build23759f5b464ad05d4eee1a2a8893d847.tmp\sketch\Ping.ino.cpp.o"
> C:\Users\Administrator\Documents\Arduino\examples\06. ?꽱?꽌?뱾\Ping\Ping.ino: In function 'void loop()':
> 
> Ping:67: error: 'microsecondsToInches' was not declared in this scope
> 
> Ping:68: error: 'microsecondsToCentimeters' was not declared in this scope
> 
> exit status 1
> 'microsecondsToInches' was not declared in this scope

 
[image]

dlabun, 2016-08-07:
If you open up the folder "C:\Users\Administrator\Documents\Arduino\examples\" outside of Arduino do the folder names show properly or do they have question marks in them? I've seen this before working with other files with Korean names.

facchinm, 2016-08-10:
@matthijskooijman can it be a side effect of https://github.com/arduino/arduino-builder/pull/156 ?
@cmaglie , should we refactor the fact that the name of the folder becomes the menu item so we don't end up with paths like `C:\Users\Administrator\Documents\Arduino\examples\06. ?꽱?꽌?뱾\Ping\Ping.ino` ?

matthijskooijman, 2016-08-10:
@facchinm Possibly, perhaps the lack of escaping previously masked problems in ctags with special characters? I guess it would be interesting to see the ctags input and output here.

dlabun, 2016-08-10:
@facchinm @matthijskooijman  Since the OP hasn't responded I am going to throw the korean language pack on my PC and see if I can get more information... I have a theory I want to test out as I have seen this issue before in other software that couldn't handle Korean, Japanese or  Chinese properly.

matthijskooijman, 2016-08-10:
@dlabun, thanks for diving into this! :-D

dlabun, 2016-08-12:
Gentlemen, I do not believe the OP's installation is a "standard" Arduino install. It looks like somebody attempted to translate the sketches into Korean... From looking at the screenshot he attached it's clear that's not the normal Ping example, it's been translated into Korean. If you try to save or open files with Korean characters the IDE throws an error stating only ASCII characters are support for file / folder names. The IDE works perfectly fine if I set the language to Korean but use only ASCII for the actual code and file names.

oqibidipo, 2016-08-23:
Minimal sketch to reproduce the bug:
```
void setup() {
  foo();
}

void loop() { }

void foo() { }
```
IDE 1.6.11 on OSX 10.11.5, sketch file name is
`~/Documents/Arduino/bug/äbç/sketch_aug23b/sketch_aug23b.ino`

>  I guess it would be interesting to see the ctags input and output here.

There is not much to see, `ctags_target_for_gcc_minus_e.cpp` is empty.

If I run the commands manually, I get:
`g++ -E` output
```
<snip>
# 24 "/Users/rjl/Applications/Arduino/Arduino-1.6.11.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs/pins_arduino.h" 2
# 250 "/Users/rjl/Applications/Arduino/Arduino-1.6.11.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h" 2
# 2 "/var/folders/sl/3x4zr7v14j5d_8xznyys7tjc0000gn/T/build0a6a8887a45148b65601f0a6c8c30d3b.tmp/sketch/sketch_aug23b.ino.cpp" 2
# 1 "/Users/rjl/Documents/Arduino/bug/äbç/sketch_aug23b/sketch_aug23b.ino"
# 1 "/Users/rjl/Documents/Arduino/bug/äbç/sketch_aug23b/sketch_aug23b.ino"
void setup() {
  foo();
}

void loop() { }

void foo() { }
```

`ctags` output
```
<snip>
setup   /Users/rjl/Documents/Arduino/bug/äbç/sketch_aug23b/sketch_aug23b.ino    /^void setup() {$/;"    kind:function
   line:1  signature:()    returntype:void
loop    /Users/rjl/Documents/Arduino/bug/äbç/sketch_aug23b/sketch_aug23b.ino    /^void loop() { }$/;"   kind:function
   line:5  signature:()    returntype:void
foo     /Users/rjl/Documents/Arduino/bug/äbç/sketch_aug23b/sketch_aug23b.ino    /^void foo() { }$/;"    kind:function
   line:7  signature:()    returntype:void
```

cmaglie, 2016-12-19:
This issue has been solved in #5277 that is basically a duplicate of this one.

closed 2016-12-19 by cmaglie