arduino/Arduino #5218: Upload fails in 1.6.10 - missing dll

General-Fault, 2016-08-07:
Executing an upload fails with the text "An error occurred while uploading the sketch". There is no other indication of the cause of the issue even with verbose upload mode enabled.
Running avrdude from the command line however indicates that "The program can't start because libsub0.dll is missing from your computer."
Verified that with version 1.6.9 the dll is present in hardware/tools/avr/bin, however in version 1.6.10 it is missing. It is still found in the application root directory.

Workaround: Copy libusb0.dll from the Arduino IDE root directory into hardware/tools/avr/bin,

General-Fault, 2016-08-07:
Didn't see issue #5189 before I submitted this. This appears to be a duplicate.

matthijskooijman, 2016-08-08:
Indeed, so I'm closing this as a duplicate.

closed 2016-08-08 by matthijskooijman