arduino/Arduino #5221: Flag to inhibit the PreProcessor from generating Function Prototypes

stickbreaker, 2016-08-08:
The subject says it all,  Is there a way to mark a function such that the PreProcessor will not generate a prototype?

It always catches me when I try to pass a complex variable as a parameter.  The Compiler generates a bogus error message about 'function declared void'.  And the line number where the error was generated usually points to my program description comments.

I know how to solve the error, Place a function prototype in a .h file, but it would be easier to somehow add a declaration or comment that controls the PreProcessor.  

I have heard the reasoning for this design decision, but I disagree with it.  It is just a scoping issue.  If beginning programmers don't understand that a function cannot be called until it has been defined.  How can they be expected to write a functioning program?

Chuck.

stickbreaker, 2016-08-14:
I was hoping this would engender some comments, but alas it appears I am an outlier.

closed 2016-08-14 by stickbreaker

cmaglie, 2016-08-14:
You can create new tabs and name the files with the extension `.cpp`. Those files are compiled as is and are not preprocessed. Anyway you must keep an empty `.ino` file with the same name of the sketch, this is compulsory (you can leave it empty or add some comments for documentation purposes).

stickbreaker, 2016-08-14:
Excellent, Thanks.