arduino/Arduino #5222: Rebuild Arduino by native GCC??

jiapei100, 2016-08-09:
Hi, is there a documentation about how to manually build and install Arduino from scratch? I mean, if I really do **NOT** want to just type install under the folder **Arduino/build/linux/dist** directly? As I can see from the current git version of Arduino, it comes with gcc 4.8.3? 
`gcc-arm-none-eabi-4.8.3-2014q1-linux32.tar.gz.sha`
I would like to build it with my gcc 5.4.0 ...

Thank you

jiapei100, 2016-08-09:
Found it.... Sorry...
[https://github.com/arduino/Arduino/wiki/Building-Arduino](url)

Chris--A, 2016-08-09:
I might be a little confused as to what you are asking, however, the compiler you linked is a cross compiler for the Due. It isn't for building the IDE.
The IDE is written in Java.

jiapei100, 2016-08-09:
@ @Chris--A  Hi, Chris, Thank you very much for your prompt reply.
Actually, my ONLY question is: **how to build Arduino from source to have it installed**? 
It looks like https://github.com/arduino/Arduino/wiki/Building-Arduino is the way out.
I actually DO NOT want to build the IDE, but only Arduino libraries, which is to be used in my C++ application.....

Is https://github.com/arduino/Arduino only for Arduino IDE? Then, which is for Arduino libraries? Thank you...

PaulStoffregen, 2016-08-09:
> I actually DO NOT want to build the IDE, but only Arduino libraries

The IDE automatically builds the libraries when you click Verify or Upload.  You can enable verbose info using File > Preferences, which will show you the exact gcc command lines the IDE is using.

This issue tracker is for reporting problems and suggesting/contributing features to the Arduino software.  The place to ask more general "how to" questions is the Arduino Forum.  Please don't distract the Arduino developers with off-topic issues here.

NicoHood, 2016-08-09:
You cannot recompile Arduino libraries like this, as they are always recompiled when you press verify/compile. The latest IDE uses GCC 4.9.x with LTO which is very good optimized and that is also what atmel uses at the moment. Even 4.8 is very new, so you do not miss much features.

If you really want to have the latest gcc compiler (for what reason ever) you want to use archlinux with makefile and not arduino. Because the few advantages the newer compiler gives mostly does not apply to AVR. Also I noticed that in some cases, especially bootloaders, gcc >4.8 produce bigger code, also combined with LTO. 

To conclude, just use the latest IDE and you are good to go and always compatible. If you want the bleeding edge write a proper makefile and use Arch. I am very happy how up to date the IDE tools are now, they've improved a lot.

matthijskooijman, 2016-08-17:
I think this question is answered, so I'm closing this issue.

closed 2016-08-17 by matthijskooijman