arduino/Arduino #5223: MKR1000 port not showing on IDE, MacOS X

michalisul, 2016-08-09:
Serial port for MKR1000 is not showing on Arduino IDE (1.6.10 and 1.6.9). The board is visible in the system itself. The cable has been used to program other boards without any problem. System version: 10.11.6, MacBook Pro (Retina, 13-inch, Early 2015).

Regards,

Michal Sulkiewicz

facchinm, 2016-08-11:
Hi Michal,
how is the board recognized in the system? Does the USB tab under "Hardware Information" shows no error? 
If you double tap the reset button to enter bootloader does the system recognize it?

michalisul, 2016-08-11:
Dear Martino,

Thank you for your quick answer. Double clicking reset button did work! The board shows in IDE now. Sorry I must have missed that information when reading your “Getting started”. For the record though I include the screenshot from USB hardware tab.

Best Regards,

Michal 

> 
> Hi Michal,
> how is the board recognized in the system? Does the USB tab under "Hardware Information" shows no error? 
> If you double tap the reset button to enter bootloader does the system recognize it?

============================
“Do or do not... there is no try.” Yoda

facchinm, 2016-08-11:
Hi Michal, I'm glad it worked.
Thanks for reporting

closed 2016-08-11 by facchinm