arduino/Arduino #5224: Add my ArduinoSTL Library.

mike-matera, 2016-08-10:
Hello Arduino Maintainers, 

I teach a C++ class using Arduino. I want my students to have the full C++ experience with vectors, cin and cout, etc so I've ported uClibc++ and packaged it as a library. This STL implementation is surprisingly functional. Using it requires about 25% of your code space on an Uno. You already have an implementation of the STL by John Wellbelove but it is broken and seems to be unmaintained (the URL in the library is a dead link). Would you please add mine? I'll maintain it! 

https://github.com/mike-matera/ArduinoSTL

Thanks!

cmaglie, 2016-08-10:
Done!

closed 2016-08-10 by cmaglie

matthijskooijman, 2016-08-10:
@mike-matera, did you also see https://github.com/maniacbug/StandardCplusplus and related forks? That also seems a port of uclibc++. Does your port add anything over that library?

mike-matera, 2016-08-10:
@matthijskooijman, Yes. I have seen that project. That one also seems to be unmaintained. If you look at the pull requests you can see that there's one from me (#17) that makes blocking I/O work properly on Arduino. It's a year old at this point. The last commit was in 2013. 

My library has two improvements going for it: It's a compliant Arduino library and it contains Andy Brown's stream implementation (in /src/serstream) that wraps a HardwareSerial in a basic_streambuf enabling cin and cout to work seamlessly.

Thanks!

matthijskooijman, 2016-08-10:
@mike-matera, true, I might see about switching over to your library :-)

Did you by chance have had contact with uclibc++ upstream? It seems some development is still going on, but I haven't found a forum, IRC channel or mailing list where development is discussed so far.

mike-matera, 2016-08-10:
@matthijskooijman. Thanks!

I didn't see a way to contact them either. The main page still lists Garrett Kajmowicz as the author but it looks like Bernhard Reutner-Fischer has been doing the bulk of the commits for years. My repo is a "rearranged" version of theirs, which keeps their commit history. My hope is that I'll be able to keep my version up to date with theirs. If you find a way to hail them I'd love to give them a hat tip.