arduino/Arduino #5230: ld error

zenz, 2016-08-11:
In 1.6.9, everything goes well.
But after upgrade to 1.6.12, when I try to modify some defines, it will report ld errors, such like while I compile the Repetier-firmware for MEGA2560:
#define EEPROM_MODE 0 => #define EEPROM_MODE 1

and then in 1.6.12, it reports:

`stepperWait' referenced in section `.text' of /var/folders/kq/lghwvplj5196rlpkyfzcgsk40000gn/T//ccreJDhi.ltrans4.ltrans.o: defined in discarded section `.text' of sketch/HAL.cpp.o (symbol from plugin)
`stepperWait' referenced in section `.text' of /var/folders/kq/lghwvplj5196rlpkyfzcgsk40000gn/T//ccreJDhi.ltrans4.ltrans.o: defined in discarded section `.text' of sketch/HAL.cpp.o (symbol from plugin)
`stepperWait' referenced in section `.text' of /var/folders/kq/lghwvplj5196rlpkyfzcgsk40000gn/T//ccreJDhi.ltrans4.ltrans.o: defined in discarded section `.text' of sketch/HAL.cpp.o (symbol from plugin)
`stepperWait' referenced in section `.text' of /var/folders/kq/lghwvplj5196rlpkyfzcgsk40000gn/T//ccreJDhi.ltrans4.ltrans.o: defined in discarded section `.text' of sketch/HAL.cpp.o (symbol from plugin)
`stepperWait' referenced in section `.text' of /var/folders/kq/lghwvplj5196rlpkyfzcgsk40000gn/T//ccreJDhi.ltrans4.ltrans.o: defined in discarded section `.text' of sketch/HAL.cpp.o (symbol from plugin)
`stepperWait' referenced in section `.text' of /var/folders/kq/lghwvplj5196rlpkyfzcgsk40000gn/T//ccreJDhi.ltrans4.ltrans.o: defined in discarded section `.text' of sketch/HAL.cpp.o (symbol from plugin)
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

while I rollback to 1.6.9, every thing ok. should be a bug?

matthijskooijman, 2016-08-11:
Could you provide a (preferably as small as possible) sketch that reproduces the problem, and/or specific links to the libraries needed?

I suspect this is caused by LTO somehow messing up, or perhaps some problem in the sketch that wasn't exposed with an earlier compiler and/or without LTO.

zenz, 2016-08-11:
https://www.repetier.com/firmware/v092/index.php

configure and download the firmware from this link.

Compile the sketch and then manually modify the some defines in Configuration.h, like above sample

#define EEPROM_MODE 0 => #define EEPROM_MODE 1

Then you can  see the error report. @matthijskooijman

Thanks

zenz, 2016-08-11:
I found the reason.

https://forum.repetier.com/discussion/2389/stepperwait-referenced-in-section-text-compiliation-issues

matthijskooijman, 2016-08-11:
I had a look at the code, and AFAICS, unlike what the library authors indicate, this is a bug in the code, not the compiler.

I downloaded the firmware from the link, and could not reproduce the problem. I suspect a workaround has been added to the code since you tried it. When I removed the `used` attribute from the stepperWait variable, and removed the `stepperWait = 0` attribute, the problem re-appears.

Looking at the code with those changes, the stepperWait variable is *only* referenced from within inline ASM blocks. However, it does so by just using the variable name, not using the input/output operand feature of the asm block. For more info, see https://gcc.gnu.org/onlinedocs/gcc/Extended-Asm.html

That same page also says:

> Accessing data from C programs without using input/output operands (such as by using global symbols directly from the assembler template) may not work as expected. Similarly, calling functions directly from an assembler template requires a detailed understanding of the target assembler and ABI.

> Since GCC does not parse the assembler template, it has no visibility of any symbols it references. This may result in GCC discarding those symbols as unreferenced unless they are also listed as input, output, or goto operands. 

Which is pretty much exactly this problem.

I'm closing this issue, but could you the Repetier maintainers to this issue so they can fix their problem properly?

closed 2016-08-11 by matthijskooijman

zenz, 2016-08-11:
OK, I will send them the message.