arduino/Arduino #5231: Makey Makey can't be programmed

bobbypop999, 2016-08-11:
When I try to remap the keys on the Makey Makey it shows this:

Arduino: 1.6.10 (Mac OS X), Board: "SparkFun MaKey MaKey"
Board makeymakey (platform avr, package SparkFun) is unknown
Error compiling for board SparkFun MaKey MaKey.

I managed it once before and it worked and I have re-installed the software and it still doesn't work.

matthijskooijman, 2016-08-11:
Can you enable verbose compilation in the preferences and post the full output here? How did you install the makeymakey board files? Did you use the board manager?

bobbypop999, 2016-08-12:
I have updated the problem please sort it.

facchinm, 2016-08-12:
This problem usually appears when you have two versions of the same core installed (due to Board Manager issue). Navigate to `/Users/$yourusername/Library/Arduino15/packages/Sparkfun/hardware/avr` and you should find at least two folders. Delete the oldest one and you should be ok. If there is only one folder the problem is somewhere else and we should try to sort it out in another way (maybe via https://github.com/arduino/arduino-builder/pull/84)

bobbypop999, 2016-08-12:
Thank you it now works and I know what to do naw if it happens again.

closed 2016-08-12 by bobbypop999