arduino/Arduino #5232: IDE does not place board info in INO file

oldmanegan, 2016-08-11:
This is a persistently frustrating issue. I searched the issues list and did not see it, my apologies if this is a repeat of another's input.
Issue: The INO file is defined to not be board/MCU specific, great; however, when working on a project involving multiple MCUs with multiple INO files open, this flexibility is a severe drawback.
Use Case: Developing a Smart Home app involving 3 systems - one with a 328, one a 1284 and one a 32u MCU (different MCU to meet a different need). I am troubleshooting the final programming/operation and running into the need to stop and switch boards and related info when switching to a INO for a specific system. A pain.
Request: Why can't I set a Preference in IDE to "stamp" the start of an INO file with the related board data to automatically be loaded and set by IDE when a specific INO has the focus? Or upon a specific key press? IDE optionally should be able to be told to ignore as well...
Requirement: Board data must include board type, clock (the 1284 is running at 20 MHz), and JTAG info.
Let me know if you need more info...

q2dg, 2016-08-12:
Well, this is an old suggestion...I've only found this issue https://github.com/arduino/Arduino/issues/5121 , but I remember having read this many times (maybe in forum).

oldmanegan, 2016-08-26:
I like the concept of making each sketch "separated" regarding each one's related hardware, instead of the hardware being a global setting in the IDE. If a "Window" menu item is created to enable ease of selecting the active project sketch (since tabs are window/project specific), then in the window domain information stored by the IDE could be the last selected hardware for that project sketch window. Make sense? Would "kill 2 birds with one stone" give a fast way to move between open sketches/windows and retain hardware info per.

per1234, 2017-03-21:
Duplicate of #1481

closed 2017-07-01 by per1234