arduino/Arduino #5234: Remote IP on EthernetClient

DarioDaF, 2016-08-11:
I sudgest adding a function to read the IP of the Clients

Changes needed:

EthernetClient.h
```
class EthernetClient : public Client {

public:

  virtual uint8_t remoteIP(uint8_t *buf); // Change

};
```

EthernetClient.cpp
```
uint8_t EthernetClient::remoteIP(uint8_t *buf) {
  if (_sock != MAX_SOCK_NUM)
    return W5100.readSnDIPR(_sock, buf);
  return 0;
}
```

It is a simple and usefull change in the library that can be used to retrive the IP address of a client connected to the EthernetServer.

PaulStoffregen, 2016-08-11:
If this is added, it probably ought to use the IPAddress class from the core library rather than an array of bytes.  Long-term, the C++ class can be much more easily extended to support IPv6.

q2dg, 2016-08-12:
Well...maybe it's worthy read this: https://github.com/arduino/Arduino/pull/1700

On the other hand, a week ago I suggested more some love to this and others improvements on Ethernet library...: https://github.com/arduino-libraries/Ethernet/issues/36#issuecomment-238085099

per1234, 2016-08-12:
> Well...maybe it's worthy read this: #1700

Yes, this was discussed in 2013 and rejected but maybe worth revisiting? Note that implementation uses IPAddress: https://github.com/arduino/Arduino/pull/1700/commits/ca37de4ba4ecbdb941f14ac1fe7dd40f3008af75
I've been using that function for a couple years and I think it would be nice for consistency with EthernetUDP which already has `remoteIP()`.

Here's the associated discussion on the Developers mailing list: https://groups.google.com/a/arduino.cc/forum/#!searchin/developers/remoteIP/developers/LW2xYj3LPM0/orkaW_SiNpAJ. The link to it on that PR has gone dead.

per1234, 2018-07-26:
Fixed by https://github.com/arduino-libraries/Ethernet/commit/ac843291454f522cdef6bddbacce2c52028be943

closed 2018-07-26 by per1234