arduino/Arduino #5236: Rebooting issue with variadic arguments, zero-pointer execution

tekka007, 2016-08-12:
AVR crashes / reboots with this sketch, asm hints to a zero-pointer call.

```
// 20160812 tekka
// rebooting issue due to zero-pointer execution

// (un)comment
#define issue

void before(void) __attribute__((weak));

void va_pseudo(uint8_t flag, ...) {
	va_list args;
	va_start(args, flag);
	va_end(args);
}

int main(void) {
	Serial.begin(115200);
  Serial.println("start");
  Serial.flush();
  
	
#if defined(issue)
	va_pseudo(1, 2, 3, 4);
#else
	va_pseudo(1, 2, 3);
#endif
 
	if (before) {
    Serial.println("This should not execute");
    Serial.flush();
		before();
	}

	Serial.println("end");
  Serial.flush();
  
	return 0;
}
```

matthijskooijman, 2016-08-16:
At first glance, the code looks sane. What does it print?

Reboot does indeed suggest zero call or jump, so the call to "before" seems the most likely place for this. I assume that removing the va_pseudo call also prevents the issue from happening, which means that somehow the vararg function messes something up that prevents the before check from happening correctly.

This sounds like a gcc/compiler issue. Did you try with both 1.6.9 and 1.6.10? (the latter has a newer gcc version)?

Thanks for reducing the example, this is something that should be reproducible from here (no Arduino around to try now, though).

tekka007, 2016-08-16:
This is the output with both Arduino 1.6.9 and 1.6.10:

#define issue**:

```
start
This should not execute
start
This should not execute
start
This should not execute
start
This should not execute
...
```

//#define issue commented:

```
start
end
```

tekka007, 2016-08-16:
ASM output:

#define issue:

```
if (before) {
 10c:	8d b7       in	r24, 0x3d	; 61
 10e:	9e b7       	in	r25, 0x3e	; 62
 110:	07 96       	adiw	r24, 0x07	; 7
 112:	0f b6       	in	r0, 0x3f	; 63
 114:	f8 94       	cli
 116:	9e bf       	out	0x3e, r25	; 62
 118:	0f be       	out	0x3f, r0	; 63
 11a:	8d bf       	out	0x3d, r24	; 61
 11c:	80 e0       	ldi	r24, 0x00	; 0
 11e:	90 e0       	ldi	r25, 0x00	; 0
 120:	61 f0       	breq	.+24     	; 0x13a <main+0x6e>
    Serial.println("This should not execute");
 122:	66 e0       	ldi	r22, 0x06	; 6
 124:	71 e0       	ldi	r23, 0x01	; 1
 126:	84 e4       	ldi	r24, 0x44	; 68
 128:	91 e0       	ldi	r25, 0x01	; 1
 12a:	0e 94 8b 02 	call	0x516	; 0x516 <_ZN5Print7printlnEPKc>
    Serial.flush();
 12e:	84 e4       	ldi	r24, 0x44	; 68
 130:	91 e0       	ldi	r25, 0x01	; 1
 132:	0e 94 f9 00 	call	0x1f2	; 0x1f2 <_ZN14HardwareSerial5flushEv>
		before();
 136:	0e 94 00 00 	call	0	; 0x0 <__vectors>
}
```

//#define issue:

```
if (before) {
 106:	0f 90       	pop	r0
 108:	0f 90       	pop	r0
 10a:	0f 90       	pop	r0
 10c:	0f 90       	pop	r0
 10e:	0f 90       	pop	r0
 110:	80 e0       	ldi	r24, 0x00	; 0
 112:	90 e0       	ldi	r25, 0x00	; 0
 114:	89 2b       or	r24, r25
 116:	61 f0       	breq	.+24     	; 0x130 <main+0x64>
    Serial.println("This should not execute");
 118:	66 e0       	ldi	r22, 0x06	; 6
 11a:	71 e0       	ldi	r23, 0x01	; 1
 11c:	84 e4       	ldi	r24, 0x44	; 68
 11e:	91 e0       	ldi	r25, 0x01	; 1
 120:	0e 94 86 02 	call	0x50c	; 0x50c <_ZN5Print7printlnEPKc>
    Serial.flush();
 124:	84 e4       	ldi	r24, 0x44	; 68
 126:	91 e0       	ldi	r25, 0x01	; 1
 128:	0e 94 f4 00 	call	0x1e8	; 0x1e8 <_ZN14HardwareSerial5flushEv>
		before();
 12c:	0e 94 00 00 	call	0	; 0x0 <__vectors>
}

```

Where it goes wrong is here:

```
 11c:	80 e0       	ldi	r24, 0x00	; 0
 11e:	90 e0       	ldi	r25, 0x00	; 0
 120:	61 f0       	breq	.+24     	; 0x13a <main+0x6e>
```

vs

```
110:	80 e0       	ldi	r24, 0x00	; 0
112:	90 e0       	ldi	r25, 0x00	; 0
114:	89 2b       or	r24, r25
116:	61 f0       	breq	.+24     	; 0x130 <main+0x64>
``` 

The zero check 

`114:	89 2b       or	r24, r25
`
is omitted where the zero pointer execution happens.

matthijskooijman, 2016-08-17:
Agreed on your analysis, the zero check is indeed needed to let `breq` do the right thing there.

I'm not sure if it's related (I don't think so), but you are using `main()` instead of `setup()` and `loop()`.

This looks like a compiler bug to me. Could you perhaps reduce the testcase further (removing the serial prints) and compile it on its own with avr-gcc (without including any of the Arduino code)? If the problem then still occurs (from looking at the asm), this is something we can report to the gcc folks.

Also, could you try compiling with gcc 5, I think a build should be available here: https://github.com/arduino/Arduino/issues/660#issuecomment-120433193

Also, could you try building with lto disabled in 1.6.10? That means removing the -flto flags from `hardware/arduino/avr/platform.txt`.

tekka007, 2016-08-21:
I've transferred the example to Atmel studio, removed all Arduino code and using only the m328p led as indicator:

This is the code:

`
#include <stddef.h>
#include <stdarg.h>
#include <avr/io.h>

#define issue

void test(void) __attribute__((weak));

void va_pseudo(int flag,...){
	va_list ap;
	va_start (ap, flag);     
	va_end (ap);    	   
}

int main(void) {
	// m328 led port output
	DDRB = 1 << 5;
	// led off
	PORTB = 0;
	
	#if defined(issue)
		va_pseudo(1, 2, 3, 4);
	#else
		va_pseudo(1, 2, 3);
	#endif
	
	if(test!=NULL) {
		test();
	}
	// led on
	PORTB = 1 << 5;
	while(1) {}
	return 0;
}`

matthijskooijman, 2016-08-22:
Thanks for further minimizing. It seems I missed your later comments, since no notification is sent when you edit a comment (so best not do that when you add significant info). I've reproduced your issue and slightly minimized it further, and submitte it upstream here: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=77326

You say it is resolved in 6.1.0. Did you compile that yourself, or is there source of prebuilt (Linux?) binaries of avr-gcc? I couldn't find any just now (for Debian / generic Linux).

tekka007, 2016-08-22:
I've found some win binaries here: http://blog.zakkemble.co.uk/avr-gcc-6-1-0/

matthijskooijman, 2016-09-21:
Upstream fixed this issue (see the bug report for details), indicating it is fixed in 6.3+ and backported to 5.5+.

I'm closing this issue, as there isn't much Arduino can do at this point, other than upgrade to a newer version when Atmel updates its patches.

closed 2016-09-21 by matthijskooijman