arduino/Arduino #5240: Problem with setHostname When Using ArduinoOTA

mhmayyan, 2016-08-13:
Hello all,

I noticed a small problem that took me a night and day to figure out.

When I use '_' in the setHostname like below
       ArduinoOTA.setHostname(XXX_YYY);
Then Arduino IDE doesn't show the device in the " Tools > Port " drop down list.

When I change it to 
       ArduinoOTA.setHostname(XXX-YYY);
Then it works.

Thanks,

per1234, 2016-08-13:
Are you referring to the [esp8266 core](https://github.com/esp8266/Arduino)'s ArduinoOTA library(https://github.com/esp8266/Arduino/tree/master/libraries/ArduinoOTA)?

mhmayyan, 2016-08-13:
@per1234 Yes I am referring to the esp8266 core's ArduinoOTA library(https://github.com/esp8266/Arduino/tree/master/libraries/ArduinoOTA).

sandeepmistry, 2016-08-15:
Closing as it's not related to this repo. I would suggest opening it here: https://github.com/esp8266/Arduino/issues

mhmayyan, 2016-08-15:
Ir is NOT a library problem. It is Arduino IDE problem.

sandeepmistry, 2016-08-15:
This might be related to https://github.com/jmdns/jmdns/issues/78.

oqibidipo, 2016-08-15:
An underscore is not a valid character in a host name.

[Wikipedia: Hostname §Restrictions on valid host names](https://en.wikipedia.org/wiki/Hostname#Restrictions_on_valid_host_names)

closed 2019-02-19 by mhmayyan