arduino/Arduino #5243: Add info about Arduino Access (chmod)

christrille, 2016-08-13:
Hi,

It would be interesting to add the information necessary to change the access right on the device.
- **Page:** [https://www.arduino.cc/en/Guide/Linux](url)
- **Comment:** sudo chmod a+rw /dev/ttyACM0
- **Reference:** [http://arduino-er.blogspot.com.es/2014/08/arduino-ide-error-avrdude-seropen-cant.html](url)

q2dg, 2016-08-13:
What you suggest can be done more elegantly executing : sudo usermod -a -G dialout <username> , which it's said in https://www.arduino.cc/en/Guide/Linux

christrille, 2016-08-16:
> dialout , which it's said in https://www.arduino.cc/en/Guide/Linux

Hi,

Good to me.

Thanks!

Chris

*******************************************
Dr. Christophe Trille
Email: ctrille@gmail.com
*******************************************
La Información incluida en el presente correo electrónico es SECRETO
PROFESIONAL Y CONFIDENCIAL, siendo para el uso exclusivo del destinatario
arriba mencionado.
Si usted no es el destinatario del mensaje o ha recibido esta comunicación
por error le informamos que esta totalmente prohibida cualquier
divulgación, distribución o reproducción de esta comunicación, le rogamos
que nos lo notifique inmediatamente y nos devuelva el mensaje original a la
dirección arriba mencionada.

Gracias.
*******************************************
The information contained in this e-mail is LEGALLY PRIVILEGED AND
CONFIDENTIAL and is intended only for the use of the addressee named above.
If the reader of this message is not the intended recipient or have
received this communication in error, please be aware that any
dissemination, distribution or duplication of this communication is
strictly prohibited, and please notify us immediately and return the
original message to us at the address above.

Thank you
*******************************************

matthijskooijman, 2016-08-16:
Indeed, the useradd method is preferable to the "give everyone write access" method suggested here. I'm closing this issue.

closed 2016-08-16 by matthijskooijman