arduino/Arduino #5244: Timer0 Compensation

schuppeste, 2016-08-13:
Hi, 

i want to use my Arduino to measure Milliseconds.

After any problems  i have found usually 2milliseconds in1 Second Drift, against the Mega2560 Crystal clocks only 15,955MHz

So, any or Every Starter on Arduino thinks they have good Accuracy on all Librarys... its not.

It would be good to modify Timer0, so everybody could calibrate the Timings for milliseconds.

matthijskooijman, 2016-08-22:
30ppm seems about right for a typical 16Mhz crystal. If you really need more accuracy, you should probably use a 32,768kHz crystal ("watch crystal"), which are often more accurate. This is not something supported by the Arduino environment, so you'll have to dig into the AVR timers yourself (IIRC timer2 supports adding an external clock, though not all chips support both external CPU and timer clocks).

I'm not sure what workaround you are proposing exactly. Do you want to the timer to run slightly faster or slower in software after calibration? If so, I'm not sure how well that will work, since I'm not sure the crystal inaccuracy is reliable (it might be too slow at one moment and too fast sometime later, depending on temperature, voltage, age, etc.). If you really want to do this, you should probably just create a my_millis() function and apply the compensation in there.

I'm closing this issue, as I do not think this is something to fix in the Arduino core. If I misunderstood, feel free to clarify in a comment.

closed 2016-08-22 by matthijskooijman

schuppeste, 2016-08-23:
Its okay, thanks!

I create a fork then to Swap Functions of Timer2 and Timer0, so i can Use Timers0 T0-Pin With 32khz Signal. Otter way i have to solder a watch crystal for asynchronous Operation instead original crystal to a new board.