arduino/Arduino #5245: 32bit boards (Due, ZEro, etc) maximum RAM

Testato, 2016-08-14:
On all Arduino 8bit board when the Verify of code finish we receive a Maximum flash and Maximum RAM message in the consolle (so if it is to hight the ide block the upload)

Instead on 32bit board the RAM control there isn't.

`Sketch uses 8,356 bytes (3%) of program storage space. Maximum is 262,144 bytes.`

Testato, 2016-12-16:
probably you must only put the upload.maximum_data_size value on all virtual board files ?
uno.upload.maximum_data_size
https://github.com/arduino/Arduino/commit/7c87db3532155c051b902ac83c1feea78eb732ee#diff-4ead488680453362ffec01d74609315cR11

per1234, 2021-03-27:
Happily, this has now been done for Arduino SAMD Boards: https://github.com/arduino/ArduinoCore-samd/pull/503 https://github.com/arduino/ArduinoCore-samd/pull/517

Still not done for Arduino SAM Boards though.

Related:
- https://github.com/arduino/Arduino/pull/1377#issuecomment-17389773
- https://github.com/arduino/Arduino/issues/1555#issuecomment-23622158
- https://github.com/arduino/Arduino/issues/5707

Testato, 2021-03-27:
will we wait another 5 years for complete resolution? :-)

ollie1400, 2022-05-02:
I made a PR into the ArduinoCore-sam repo, basically doing what @per1234 suggested (thanks for the suggestion!).  Any comments are welcome!