arduino/Arduino #5247: Ubuntu /  Arduino IDE

VaclavSal, 2016-08-14:
Could someone  please answer these questions for me? 
A reference  / link on how to would do also. 

Downloaded latest IDE 1.6.10 Linux 64 bit version.

I have no menu option to select "Due " board.
I believe I need access to "Library Manager" to gain access to Due / ARM software.
**How  do I access "Library Manager " in Linux?**  

The IDE title version does not indicate IDE  1.6.10 running.
Does the Linux IDE versioning uses different numbering convention than Windows IDE ?
If so , how do I know I am using the latest Linux IDE release?

I am sorry to post these questions here, however, since there are no similar posts (about Linux ) on the Arduino.cc forum I had very little choice.

Thanks for understanding and I do appreciate your time.

per1234, 2016-08-15:
>I have no menu option to select "Due " board.

https://www.arduino.cc/en/Guide/Cores
>If so , how do I know I am using the latest Linux IDE release?

**Help > About Arduino** look at the top left corner of the pop up
>since there are no similar posts (about Linux ) on the Arduino.cc forum I had very little choice.

If you had spent any time searching the forum you would know it's **Boards Manager**, not Library Manager. Why would you need to find Linux specific instructions for installing Due support? Do you actually think they have completely different IDE versions for each OS?

If you're smart enough to program a Due you're smart enough to figure this stuff out on your own with a minimum of effort. You've already been told that issue reports are to report bugs in the IDE, not as your personal support line to the developers. They have much more important things to do than hold your hand every time you have a problem.

sandeepmistry, 2016-08-15:
Please see @per1234's comment above.

@per1234 thanks for answer this.

closed 2016-08-15 by sandeepmistry