arduino/Arduino #5249: Wire.endTransmission(); freezes if connection gets lost

holgerlembke, 2016-08-14:
Wire.endTransmission(); freezes, when connection somehow gets lost. A simple cable issue might let endTransmission() do its while() loops in twi_writeTo and twi_readFrom eternally.

There is no easy way an end user can detect this situation, avoid this situation or react to this situation.

As far as I read from the forums this is a lengly discussed issue. Following the discussions there already is a patch:

http://forum.arduino.cc/index.php?topic=344506.msg2374342

Could we please move it into current code base? Thanks.

facchinm, 2016-08-22:
Duplicate of https://github.com/arduino/Arduino/issues/1476 , following the conversation there

closed 2016-08-22 by facchinm

ermtl, 2020-03-27:
It's been 9 FUCKING YEARS since that bug was first discussed (2011), countless people pulled their hair trying to understand why their Arduino was freezing, why would it work normally, then suddenly stop, a dozen of times, it's been raised in here, dozens of times people had been told to get lost, use something else, etc ...
#7328
#2418
#1842
#5249
arduino/ArduinoCore-avr#42
Google shows pages after pages of people getting started and who get discouraged by this elusive bug they don't understand. This runs totally contrary of what arduino is supposed to stand for.

If the fix was difficult, if it compromised compatibility or added other problem, it would be understandable, but it's not the case, all that shit is caused by those 2 damn lines of code that obviously can create an infinite loop if for some reason the read operation does not complete :

// wait for read operation to complete
while(TWI_MRX == twi_state){
continue;
}

Yes, I know, this state is not supposed to happen according to the I2C protocol, but guess what ? electrical glitches didn't get the memo.

Countless people, after losing hours or days kind of solved the issue either by making a modified version for themselves, or switching to another library, so there are several implementations of a timeout that are simple and would easily solve the issue.

But arduino developpers stubbornly refuse to fix it !

You think I'm rude ? After 9 years of giving the finger on this issue to the whole community for absolutely no reason, I couldn't care less.

How long before you close it again ?
Just be honest, and put a "fuck you all, WONTFIX" label on it ...