arduino/Arduino #5251: IDE nightly 6.1.11 has broken library version strings in library manager

bperrybap, 2016-08-15:
The latest nightly build (Arduino 1.6.11 Hourly build 2016/08/15 07:31) seems to be truncating the library version strings after the Z point number rather than using/showing the full Semver version string.
You can see this using my hd44780 library.
Look at the available versions to install.
On 1.6.10 you can currently see versions 0.6.0 and 0.5.0-a which is correct.
However on 1.6.11 the versions are:
0.6.0 and 0.5.0 which is incorrect.
The IDE should not be truncating the version string since things after a Z point number are still a valid portion of the version string used to distinguish between versions.

See rule semver rule 9 http://semver.org/#spec-item-9 and semver rule 11 http://semver.org/#spec-item-9
which show describes additional version information beyond the Z point number and how to use it.

Full semver spec:
http://semver.org/

The library_index.json file seems to have all the correct information, so this looks like someone altered the IDE to truncate the version string displayed in the library manager.

The IDE library manager needs to show the full version string otherwise, there will be no way for a user to distinguish between multiple pre-release versions and the final version as they will all look like the same version.

bperrybap, 2016-08-15:
Also, the text boxes used to select the version to install are also broken in 1.6.11
They used to be wide enough to see the version strings, now they are not wide enough until you click on them.

The  text box used to show most recent version isn't large enough to show even a simple 1.0.0 version string.
It widens when you click on it, but there is plenty of room to show it before you click on it.
1.6.10 did not have this issue.

And for libraries that are already installed, the text box that says "Select version" is not wide enough.
All that shows is "Select versi...".
The box widens when you click on it, but there is plenty of room to show it before you click on it.
1.6.10 did not have this issue.

cmaglie, 2016-08-16:
I confirm that there is something fishy going on the nightly. I'm working on it.

cmaglie, 2016-08-16:
Thanks @bperrybap  for reporting this, the fix will be available on the next hourly build.

commit 49b98959 (2016-08-16): Correctly handle "-snapshot" and "+build" in semantic versioning

This fix a regression introduced in:

048a8a61 (VersionHelper now correctly strip snapshot info)

actually neither 048a8a61 nor the version before are correct becuase:

048a8a61 - strips all the extra `-snapshot` and `+build`
previous - doesn't handle the case `x.y-snapshot`

Now both are handled correctly and a test has been added to verify this.

To be completely semver compliant we should deny versions in the
format `x.y`, but this will break all legacy version that have been
published until now, so this changed should be pos [... cut]
closed 2016-08-16 by cmaglie