arduino/Arduino #5253: Unable to install IDE 1.6.10 in Ubuntu (16.04) from github

VaclavSal, 2016-08-16:
After following the published procedure how to install from Github  I am unable to select "Due"' as board.
.....
Open Ubuntu Software Center and search for Arduino. Alternatively, you can install via the command line by running the following in a Terminal

 
sudo apt-get update && sudo apt-get install arduino arduino-core  

The above installs a package from the Ubuntu software repositories, which currently is the old Arduino version 1.0.5. 
...

**The procedure indeed installs Arduino version 1.0.5. (?)  which does not have "Board Manager" menu option.**

cmaglie, 2016-08-16:
1.0.5 is an **ancient** version, the support for the Due has been introduced starting from 1.5.x

```
ARDUINO 1.0.5 - 2013.05.15
```

closed 2016-08-16 by cmaglie

PaulStoffregen, 2016-08-16:
You need to download the software from Arduino.cc's Download page.

The *last* thing you want is the terrible version Debian & Ubuntu publish from apt-get.  On top of being ancient, it uses a different version of the AVR toolchain, so on top or being ancient, it isn't even fully compatible with the 1.0.5 version published years ago.

VaclavSal, 2016-08-16:
sudo apt-get update && sudo apt-get install arduino arduino-core 

Thanks, 
so it looks as the command "update" is the real issue? 
Can you confirm that much?

matthijskooijman, 2016-08-16:
The issue is that the Debian / Ubuntu maintainers have not included the latest versions of Arduino in their repositories (for various reasons). For this reason, using apt-get to install Arduino is not recommended, and you are better off downloading and unzipping it directly from arduino.cc.

VaclavSal, 2016-08-16:
**downloading and unzipping it directly from arduino.cc produces SAME results** 

Here is what I see in Terminal 

jim@jim-desktop:/usr/share$ arduino
The program 'arduino' is currently not installed. You can install it by typing:
sudo apt install arduino
jim@jim-desktop:/usr/share$ sudo apt install arduino
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  arduino
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/1,164 kB of archives.
After this operation, 1,727 kB of additional disk space will be used.
Selecting previously unselected package arduino.
(Reading database ... 200611 files and directories currently installed.)
Preparing to unpack .../arduino_2%3a1.0.5+dfsg2-4_all.deb ...
Unpacking arduino (2:1.0.5+dfsg2-4) ...
Processing triggers for gnome-menus (3.13.3-6ubuntu3.1) ...
Processing triggers for desktop-file-utils (0.22-1ubuntu5) ...
Processing triggers for mime-support (3.59ubuntu1) ...
Processing triggers for bamfdaemon (0.5.3~bzr0+16.04.20160701-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for hicolor-icon-theme (0.15-0ubuntu1) ...
Processing triggers for shared-mime-info (1.5-2ubuntu0.1) ...
Processing triggers for man-db (2.7.5-1) ...
**Setting up arduino (2:1.0.5+dfsg2-4) ...**
jim@jim-desktop:/usr/share$ arduino
jim@jim-desktop:/usr/share$

matthijskooijman, 2016-08-16:
You just posted the output of apt-get, while I just explained to not use it. I don't see you're actually listening at all, so I'll not invest further time into this. Good luck!

sandeepmistry, 2016-08-16:
@VaclavSal please follow https://www.arduino.cc/en/Guide/Linux

Questions like this should be asked on the forum (http://forum.arduino.cc), Github issues is not a place for general support, only for reporting issues with the IDE and AVR + SAM cores.

VaclavSal, 2016-08-16:
I followed this process:

If you don’t find the option to run the script from the contextual menu, you have to open a Terminal window and move into the arduino-1.6.x folder. Type the command ./install.sh and wait for the process to finish. You should find a new icon on your desktop. 

It does installs new icon on the Desktop but that is all, no other software is installed. 
Executing the icon does nothing.