arduino/Arduino #5254: Boards Package 1.6.12 broken

lpdunwell, 2016-08-16:
I get a "undefined reference to `__eerd_block_m2560'", which doesn't appear with 1.6.11.
Actually, just by downgrading back to 1.6.11 and no IDE restart I get a successful compile.

cmaglie, 2016-08-18:
can you post a sketch that reproduce the problem?

lpdunwell, 2016-08-19:
I can't even reproduce the error with the exact setup i had.
fail.
closed for now.

closed 2016-08-19 by lpdunwell

oqibidipo, 2016-08-19:
Looks like a mix of old `avr/eeprom.h` and avr-libc 2.0.

Old `avr/eeprom.h`:
```
...
#elif defined (__AVR_ATmega2560__)
# define _EEPROM_SUFFIX  _m2560
...

#define _EEPROM_CONCAT1(s1, s2)     s1 ## s2
#define _EEPROM_CONCAT2(s1, s2)     _EEPROM_CONCAT1 (s1, s2)

#define eeprom_read_byte      _EEPROM_CONCAT2 (__eerd_byte, _EEPROM_SUFFIX)
#define eeprom_read_word      _EEPROM_CONCAT2 (__eerd_word, _EEPROM_SUFFIX)
#define eeprom_read_dword     _EEPROM_CONCAT2 (__eerd_dword, _EEPROM_SUFFIX)
#define eeprom_read_float     _EEPROM_CONCAT2 (__eerd_float, _EEPROM_SUFFIX)
#define eeprom_read_block     _EEPROM_CONCAT2 (__eerd_block, _EEPROM_SUFFIX)
...
```

Avr-libc 2.0 does not rename the functions.