arduino/Arduino #5256: No menu to list currently opened sketches in MacOS X

nesdnuma, 2016-08-16:
With several sketches opened simultaneously, it's a real pain to navigate between them whenever any of them is hidden. Adding a "sketches" menu listing all the currently open sketches would help a lot.

oldmanegan, 2016-08-16:
Of course, there is Command + ~ to rotate through open sketches currently, which is barely acceptable. And it seems to not follow a predefined order, instead it moves forward through the list of open sketches until you work on a sketch, then the next time it moves backward through the list. Would be nice to have the menu of open sketches along the bottom bar, as well as the open hardware configs, to make it easy on users with different sketches using different HW configs open.
Regards,

John
johnvegan@gmail.com

nesdnuma, 2016-08-16:
By the way, there's no dedicated ~ key on French azerty keyboards, so the mentioned shortcut is not even possible for me. However one can install [this little app](url=http://manytricks.com/witch/).

lmihalkovic, 2016-08-24:
See #234 
Seems it was first requested in 2010...

per1234, 2017-07-05:
Thanks for pointing that out @lmihalkovic!

Closing as duplicate of https://github.com/arduino/Arduino/issues/234

closed 2017-07-05 by per1234