arduino/Arduino #5260: Custom menu selection text doesn't update after changing boards when menu ID is different but menu text is the same as a previously selected board

per1234, 2016-08-17:
Using Arduino IDE 1.6.11 Hourly Build 2016/08/16 11:33 with Windows 7 64 bit

- File > Preferences > Additional Boards Manager URLs: > https://zevero.github.io/avr_boot/package_zevero_avr_boot_index.json
- Tools > Board > Boards Manager > avr_boot > Install > Close
- Tools > Board > Arduino/Genuino Mega or Mega 2560
- Tools > Board > ATmega328/P

Now look at the selection text shown in quotes next to the **Tools > Board > Processor** menu item: `Processor: ATmega2560 (Mega 2560)`. I you select the **Tools > Board > Processor** menu you can see that it doesn't have an `ATmega2560 (Mega 2560)` option. That text has been left over from the previous board selection. Even after changing the Processor menu selection the menu item text doesn't change.
[image]

This only occurs when the menu ID value in boards.txt is different from the Arduino AVR Boards(or any other core) value but the menu text value in boards.txt is the same as the other core's value. In this [case](https://github.com/zevero/avr_boot/blob/gh-pages/avr/boards.txt#L8):
```
menu.processor=Processor
```
If this is changed to `menu.cpu=Processor`(the [same ](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/boards.txt#L3)as Arduino AVR Boards) then the issue doesn't occur.

If it's changed to `menu.processor=CPU` then the issue doesn't occur.

If it's changed to `menu.avr_boot_processor=Processor` the issue does occur.

cmaglie, 2016-08-17:
> If it's changed to menu.avr_boot_processor=Processor the issue does occur.

this remembers me about an old issue involving underscores in the submenu definitions.
Just a blind guess: may you try to remove all underscores from the submenu name?

`menu.avr_boot_processor=Processor` -> `menu.avrbootprocessor=Processor`

per1234, 2016-08-17:
> menu.avr_boot_processor=Processor -> menu.avrbootprocessor=Processor 

I tried it with the `avrbootprocessor` MENU_ID and it still causes the issue.

cmaglie, 2016-08-17:
Ok, following your instructions I can reproduce the issue.
It seems that the menu item is not updated correctly BTW, besides the GUI issue, it seems that the selection is working correctly, can you confirm that?

Fixing this bug requires touching the submenu selection logic in the IDE (that is [quite complex really](https://github.com/arduino/Arduino/blob/master/app/src/processing/app/Base.java#L1427-L1625)) and it will not happen for 1.6.11, I'm setting the milestone for this bug to 1.6.12.

per1234, 2016-08-17:
> it seems that the selection is working correctly, can you confirm that?

Yes, it works correctly.

Thanks for looking into this. I agree it's not a high priority bug. To fix this specific case I only need to change the MENU_ID to `cpu` when I do the next avr_boot release but I've confirmed this also happens when the same conditions occur with any other installed boards packages so it's possible for another combination to cause the issue.

hasenradball, 2019-05-04:
Hi is there already an solution available?

per1234, 2019-05-04:
There is no solution yet to the bug in the Arduino IDE. However, the workaround for the issue is to change the menu label text in one of the boards.txt files. The issue only occurs when multiple boards.txt files use the same menu label text.

In the particular case of having both the ESP8266 and ESP32 cores installed, that could be done by changing line 1 of the ESP32 boards.txt from:
```
menu.UploadSpeed=Upload Speed
```
to:
```
menu.UploadSpeed=Upload Baud Rate
```
(or whatever text you prefer, as long as it's not "Upload Speed"). Of course you could change the ESP8266 boards.txt instead if you prefer.

Note that you'll need to redo this every time you update to a new version of the ESP32 core.

hasenradball, 2019-05-04:
Hi,
thanks for the fast reply, this could be an option for me.
Is it correct that I have only to change the line 1, which is the name of the menue?

per1234, 2019-05-04:
>Is it correct that I have only to change the line 1

That's correct. That line defines the menu label text for all the ESP32 boards that have an `UploadSpeed` menu.

hasenradball, 2019-05-04:
Ok Fine works perfect! :-)

I change it Upload Speed (ESP32) -> for the ESP32
and Upload Speed (ESP8266) -> ESP8266

Thanks!

[image]

PR #8880 (2019-07-18): Annotate custom menu to avoid name clashing
Fixes #5260

@per1234 @hasenradball  can you test in your environment too? Thanks!
commit 06cfbbfa (2019-07-18): Annotate custom menu to avoid name clashing

Fixes #5260
closed 2019-07-18 by facchinm