arduino/Arduino #5264: Read-only file system on Mac Os Sierra

damianvdk, 2016-08-18:
Because of the new preferences, Arduino isn't able to edit a file because it's locked with Read-Only. 
[image]
[image]

petersterling, 2016-09-15:
Yes, this issue is causing me a lot of pain at the moment too. Any ideas when it will be resolved?

damianvdk, 2016-09-15:
@petersterling yes, i think i was able to do it. But then i got another error and i gave up. I just used my old laptop then. I can send you a video on how I did it of you want?

facchinm, 2016-09-15:
We are working on a couple of Sierra-related fixes which will be published just after Sierra goes "live", so don't worry :smile:

petersterling, 2016-09-15:
Thanks @facchinm it will be good to get going again! I don't suppose there is any way to get a beta of the IDE to go along with my macOS dev beta? Cheeky, but you gotta ask!  :-)

sandeepmistry, 2016-09-15:
I don't have Sierra installed, but something for you to try out for us:

Download IDE, unzip, open once, exit. Then in the terminal app run:

```sh
cd /Applications/Arduino.app/Contents/Java/hardware/tools/avr/lib

install_name_tool -id libusb-1.0.0.dylib libusb-1.0.0.dylib
install_name_tool -id libusb-0.1.4.dylib libusb-0.1.4.dylib
install_name_tool -change /Users/jenkins/jenkins/workspace/toolchain-avr-mac32/objdir/lib/libusb-1.0.0.dylib "@executable_path/../lib/libusb-1.0.0.dylib" libusb-0.1.4.dylib

cd ../bin
install_name_tool -change /Users/jenkins/jenkins/workspace/toolchain-avr-mac32/objdir/lib/libusb-1.0.0.dylib "@executable_path/../lib/libusb-1.0.0.dylib" avrdude_bin
install_name_tool -change /Users/jenkins/jenkins/workspace/toolchain-avr-mac32/objdir/lib/libusb-0.1.4.dylib "@executable_path/../lib/libusb-0.1.4.dylib" avrdude_bin
mv avrdude avrdude.sh
mv avrdude_bin avrdude
chmod +x avrdude
```

Maybe only the `chmod` is needed, but this removes the need for the old bash script.

petersterling, 2016-09-15:
@sandeepmistry - thanks for that... But I had just downloaded the nightly build version of the Mac IDE, I guess while you were typing, and it works! While this is great news for me, is it what you would have expected?

sandeepmistry, 2016-09-15:
I just tried the hourly build, it doesn't appear to be signed like the official IDE releases, so maybe it doesn't have this same restrictions?

@facchinm any ideas on this?

facchinm, 2016-09-16:
Yes, Apple has no restrictions on unsigned apps (since you are prompted to accept to run it anyway).
I'll produce a signed app to test the correct functionality in the next couple of hours

cmaglie, 2016-09-22:
Fixed with #5374 and released with IDE 1.6.12.

closed 2016-09-22 by cmaglie