arduino/Arduino #5267: [Library Manager] Please add these libraries

ryantm, 2016-08-19:
Hi.

We are working on getting all of our libraries into the library manager! Can you please add the following libraries to the library manager?

https://github.com/pololu/fastgpio-arduino
https://github.com/pololu/dual-vnh5019-motor-shield/
https://github.com/pololu/qtr-sensors-arduino
https://github.com/pololu/qik-arduino
https://github.com/pololu/lsm303-arduino
https://github.com/pololu/lps-arduino/
https://github.com/pololu/pololu-hd44780-arduino
https://github.com/pololu/drv8835-motor-shield
https://github.com/pololu/a4990-motor-shield
https://github.com/pololu/l3g-arduino

Thanks,
Ryan

agdl, 2016-08-22:
Done thank you!

closed 2016-08-22 by agdl

ryantm, 2016-08-22:
@agdl Thank you! Did you add the DRV8835MotorShield? I don't see that when I search in the Library Manager. All of the rest are there.

`curl http://downloads.arduino.cc/libraries/library_index.json | grep DRV8835` yields no matches

agdl, 2016-08-23:
You are right! I missed it. It is there now!