arduino/Arduino #5272: [Library Manager] Request to add to library manager: Ethernet 2

njh, 2016-08-20:
The library for official Ethernet Shield 2 is not in the Library Manager:
http://www.arduino.org/products/shields/arduino-ethernet-shield-2

I assume it is this one from Adafruit:
https://github.com/adafruit/Ethernet2

There has been no response to adafruit/Ethernet2#4

Please can you add it?

per1234, 2016-08-21:
Adafruit would need to make a release or tag for that library to be picked up by Library Manager.

njh, 2016-08-21:
@toddtreece has created a tag for version 1.0:
https://github.com/adafruit/Ethernet2/tree/1.0

cmaglie, 2016-08-23:
Can you update the `maintainer` field to `Adafruit <.....>`?

toddtreece, 2016-08-23:
@cmaglie fixed with [v1.0.1](https://github.com/adafruit/Ethernet2/releases/tag/1.0.1)

cmaglie, 2016-08-29:
Hi @toddtreece,

may I ask you to change also the `author=` field?
as it is now it looks like Arduino is the author, instead it seems that this library is a work made by WizNet.

I'm sorry for being picky on this one, hope you understand my concerns.
If you can change the author field I'll add it straight away.

Thanks!

njh, 2016-08-30:
Note that Wiznet are not the author of most of the code in the library.

Based on the header files, only w5100.* is written by the Wiznet team, the majority of the files say "by Arduino.org team" or other individuals or companies.

The official Wiznet library is very different:
https://github.com/Wiznet/ioLibrary_Driver

toddtreece, 2016-08-30:
@cmaglie is `Adafruit <info@adafruit.com>` the appropriate choice for author? i wasn't sure since the lib has multiple authors like @njh mentioned. just let me know what you think and i can update the field.

cmaglie, 2016-08-30:
@njh
it seems that the origin is quite mixed up, BTW I was referring to this one: https://github.com/Wiznet/WIZ_Ethernet_Library (that in turn forks from https://github.com/Wiznet/WIZ_Ethernet_Library...)

whatever, the amount of external contributions is very high, IMHO the correct definition is: `author=Various'

Probably a better solution is:

1) add a CONTRIBUTIONS file with the list of all contributors
2) set `author=Various (see CONTRIBUTIONS file for details)`

The same must be done for the `Ethernet` library. I'll open an issue for that.

BTW for now just set `author=Various`, and move on :-)

toddtreece, 2016-08-30:
@cmaglie thanks for the feedback. i added an issue to the lib for adding `CONTRIBUTIONS`. here is the updated release: https://github.com/adafruit/Ethernet2/releases/tag/1.0.2

cmaglie, 2016-08-30:
Done!

closed 2016-08-30 by cmaglie

njh, 2016-08-31:
Thanks!

http://www.arduinolibraries.info/libraries/ethernet2