arduino/Arduino #5273: [DOC] https://www.arduino.cc/en/Reference/SPI

olikraus, 2016-08-21:
The table about the SPI Modes is wrong for modes 2 and 3. Additionally i suggest to add a column on the clock edge for the data takeover of the client device. 

| Mode	| Clock Polarity (CPOL)	| Clock Phase (CPHA)	| Output Edge  | Data Takeover | 
|---|---|---|---|---|
| SPI_MODE0	| 0	| 0	| Falling |  Rising | 
| SPI_MODE1	| 0	| 1	| Rising |  Falling | 
| SPI_MODE2	| 1	| 0	| **Rising** | Falling  | 
| SPI_MODE3	| 1	| 1	| **Falling** | Rising | 

References:
Forum discussion:
http://forum.arduino.cc/index.php?topic=419660.0

Existing issues related to this topic:
olikraus/u8g2#52
esp8266/Arduino#2416

agdl, 2016-08-22:
Done thank you!

closed 2016-08-22 by agdl

olikraus, 2016-08-23:
Thanks
Unfortunately there is a small spelling mistake on the web page (marked in bold below)

| Mode	| Clock Polarity (CPOL)	| Clock Phase (CPHA)	| Output Edge  | Data Capture | 
|---|---|---|---|---|
| SPI_MODE0	| 0	| 0	| Falling |  Rising | 
| SPI_MODE1	| 0	| 1	| Rising |  Falling | 
| SPI_MODE2	| 1	| 0	| Rising | **Flling**  | 
| SPI_MODE3	| 1	| 1	| Falling | Rising |

agdl, 2016-08-23:
Fixed sorry for that