arduino/Arduino #5276: 1.6.11 windows: subfolder include breaks build?

i-make-robots, 2016-08-21:
sorry if this is a duplicate.  I tried to search first!

I am running 1.6.11.  
I am compiling https://github.com/MarginallyClever/Makelangelo-firmware/blob/master/firmware_ams/firmware_ams.ino
/firmware_ams/ has a subfolder AFMotorDrawbot with AFMotorDrawbot.h and AFMotorDrawbot.cpp.

When I compile, I get errors about the contents if AFMotorDrawbot.*.

If I move AFMotorDrawbot.* to /firmware/ (up one directory) and change the include path everything compiles OK.

matthijskooijman, 2016-08-22:
In 1.6.10, the recursive compilation of sketch files was limited to the src/ subfolder. Try creating a firmware_ams/src folder and moving your subfolder into there, I believe it will then be included in the compile again (though I'm not sure I understand the problem entirely, some info is missing).

Also, this report tracker is meant for bugs in the IDE, not for help with getting your code compiled. This particular problem looks like it might be a bug, so it's a bit of an edge case, but it might have been better asked on the forums first (and coming here if the conclusion is that it is indeed an IDE bug).

i-make-robots, 2016-08-22:
FWIW I asked in IRC #arduino first.  In any case, you admit it is the
correct place.  So.

why was the normal behavior of #include changed to work only in the src/
folder?  I have sub and sub-sub folders that cannot all be smashed into one
src/ folder - the naming conflicts would be bad.

per1234, 2017-07-05:
Closing as duplicate of https://github.com/arduino/Arduino/issues/5176.

per1234, 2017-07-05:
By the way, @i-make-robots you don't need to put all the files directly in the `src` folder. You are welcome to use subfolders of the `src` folder to keep things separated.

closed 2017-07-05 by per1234

i-make-robots, 2017-07-05:
@per1234 I have been doing that for quite some time.  When I ran 1.6.11 the behavior changed and it refused to agree with you.