arduino/Arduino #5277: Accent and Tab's ERROR

danilodene, 2016-08-21:
In the version 1.6.10 and 1.6.11 folder that has accent the IDE doesn't recognition TAB's

Error: functionXXX was not declared in this scope

matthijskooijman, 2016-08-22:
Could you provide a complete (but minimal) example showing this problem? A sketch in zipfile is probably easiest, or just including the code in a comment (using the ``` code block markdown syntax around the code)

danilodene, 2016-08-23:
Sorry, I uploaded the video demonstrating the problem:  https://www.youtube.com/watch?v=AMB4zOs6bRs&feature=youtu.be

oqibidipo, 2016-08-23:
Seems to be the same problem as in #5216.

dlabun, 2016-08-23:
It's the same issue, non-ASCII characters don't work in the IDE. The IDE throws error messages when you try to open / save files from the IDE with non-ASCII characters but it doesn't give you any useful errors during compiling. The only real fix to this would be enable full Unicode support.

facchinm, 2016-08-26:
The problem is due to https://github.com/arduino/arduino-builder/pull/156/files#diff-8f49a7d3dafee03fdad8aeee774529afR426
The `ParseCppString` function doesn't take into account the fact that a character could be a multibyte rune.
Ugly solution [here](https://gist.github.com/facchinm/07db68e35e419bc5fef3a190ad6d7528), needs to be refactored/understood well but it works (the code was borrowed from https://github.com/golang/go/blob/master/src/net/mail/message.go)

@matthijskooijman

dlabun, 2016-08-26:
@facchinm Until a full solution can be implement could an error message be added in? At least it could explain the issue to users.

facchinm, 2016-08-29:
@dlabun , it should be solved in the next few days, then it will be merged in nightly build so no need to add strings to explain something that is going to be fixed soon :wink:

facchinm, 2016-08-29:
@dlabun , in the meantime you could test this PR https://github.com/arduino/arduino-builder/pull/178#issuecomment-243080177

Once you download the linked zip, replace the `arduino-builder` executable you can find in the IDE installation folder with the provided one

oqibidipo, 2016-08-29:
My example from 5216 compiles without errors, also with a directory name like <smiley-face><lion-face><thunder-and-rain><snowman>.

dlabun, 2016-08-29:
@facchinm Will do, should have results back this evening. I am going to test with Korean characters as well so #5216 can closed as well.

closed 2016-11-23 by facchinm