arduino/Arduino #5279: boolean invert bug

joahend, 2016-08-22:
boolean blink=false;

void loop() {
    blink=!blink;
    delay(200);
    Serial.println(blink);
}

work fine until version 1.6.9
doesn't work on versions 1.6.10 and 1.6.11

tested with nano

matthijskooijman, 2016-08-22:
> i cant say what part of rest of code affect this routine

Please try reducing the code further, removing all other code, and see if it still breaks. If so, please post the *full* sketch that shows the behaviour. Also, please be more specific about what the resulting behaviour is (your comment says "always only false", but that line should, AFAICS, always print 0/1, not true/false anyway...).

sandeepmistry, 2017-02-16:
Closing this for now due to lack of activity.

closed 2017-02-16 by sandeepmistry