arduino/Arduino #5281: EspDuino bord not working with 1.6.11

angipp01, 2016-08-23:
Once installed 1.6.11  The board EspDuino is not working any longer.  

C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Arduino.h:279:26: fatal error: pins_arduino.h: No such file or directory

 #include "pins_arduino.h"

                          ^

compilation terminated.

exit status 1
Error compiling for board ESPDuino (ESP-13 Module).

angipp01, 2016-08-23:
Also ESPDuino board works fine with Ver 1..6.8 houry build 2016/0105 04:00

per1234, 2016-08-23:
Also posted on the Arduino forum: http://forum.arduino.cc/index.php?topic=419762

Testato, 2016-08-26:
ESPduino is not an official Arduino product, so you must write on the author Github. He must update ESPduino for 1.6.11 IDE

cmaglie, 2016-08-26:
The fact that pins_arduino.h is not found sound like a possible problem in the Arduino IDE, I didn't had the time to look in detail into this issue yet.

per1234, 2016-08-26:
@angipp01 managed to get things working(see forum thread) but hasn't been kind enough to respond to my questions to make it easier for me look into the actual cause of the issue. So I had to spend a bunch of time searching around to find what was going on.

ESPduino is one of the [esp8266 core](https://github.com/esp8266/Arduino)'s boards in the latest version.

As you can see from the path, @angipp01 is using an old version, 2.0.0, of the core. When I installed that version I was surprised to see that ESPduino is not one of the boards. After a bunch more searching I found the reason why @angipp01 was able to use the ESPduino board with esp8266 core 2.0.0 and also why they were using the "Ver 1..6.8 houry build 2016/0105 04:00" of the Arduino IDE. It turns out that the manufacturer of the ESPduino board, "Doctors of Intelligence & Technology" have a full version of the Arduino IDE with the esp8266 core installed and modified with their ESPduino board(http://en.doit.am/arduino1.6.8-ESPDuino-2.1.zip) available for download. This installation runs in portable mode and you can compile for the ESPduino board without errors.

Their custom Arduino IDE version comes with the Additional Boards Manager URL in preferences: http://espduino.doit.am/package_esp8266com_index.json. Failing to live up to their name, the Doctors forgot to add the espduino variant to the v2.0.0 file at the URL given in that package(http://cn.doit.am/2.0.0/esp8266-2.0.0.zip) so if you use Boards Manager to install it with any Arduino IDE version you get the error @angipp01 reported, as is correct since that file doesn't exist.

@angipp01 left the conversation with the cryptic message:
> thanks that it!..  no more errors.  thanks
http://arduino.esp8266.com/versions/2.3.0/package_esp8266com_index.json

My interpretation of this is that they changed the URL to http://arduino.esp8266.com/versions/2.3.0/package_esp8266com_index.json, installed the official esp8266 core's v2.3.0 and the problem was fixed. Unfortunately for @angipp01 that URL is also not the best option since it won't provide updates beyond 2.3.0 but I guess that's a just punishment for abandoning the issue.

I wouldn't spend any time investigating this issue @cmaglie.

cmaglie, 2016-09-15:
thanks @per1234

closed 2016-09-15 by cmaglie