arduino/Arduino #5283: [Library manager] UsbHost library always appears as updatable

akash73, 2016-08-23:
the library manager shows me UsbHost library as updatable even if is the latest available.
Tried to downgrade to 1.0.4 and back to 1.0.5 but didn't resolve.

IDE 1.6.11 OSX 10.10.5

[image]

per1234, 2016-08-23:
I can reproduce this with any SAMD board selected (using Arduino IDE 1.6.11 Hourly Build 2016/08/22 03:34 with Arduino SAMD Boards 1.6.6). It does not occur with any Arduino AVR Boards or Arduino SAM Boards board selected.

I think the problem is caused because Library Manager sees that a newer version of the library(1.0.5) is available than the version included with Arduino SAMD Boards(1.0.0). However, when you download the version 1.0.5 the `architectures` value is `sam` so it still sees that 1.0.0 is installed. Obviously there is some logic missing from Library Manager and a similar issue was reported quite some time ago(https://github.com/arduino/Arduino/issues/4189) and wrongly closed. I tried to get it reopened without success.

The solution I've been using with 3rd party hardware packages, where this frequently causes problems, is to change the `name` value in library.properties to a unique value(you don't need to change any filenames). In this case it could be:
```
name=USBHostI(Arduino SAMD Boards)
```
This solves the problem and actually makes sense because these are two different libraries, why should they have the same name?

There does seem to be a new version of this issue with the same behavior but a different cause that appeared in Arduino IDE 1.6.10(encountered with the FreeRTOS library in https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/196). I had put together a demonstration of this but decided since I've only seen it affect 3rd party hardware packages and can easily be fixed in those packages that it was better not to bother the Arduino developers with it. If this is something you want to look into I can check if it's still occurring and give more information.

per1234, 2016-08-23:
I just had a look at the version of the library at https://github.com/arduino-libraries/USBHost. The readme says:
> The USBHost library allows an Arduino Due and Zero board to appear as a USB host,

This indicates the solution is as simple as changing https://github.com/arduino-libraries/USBHost/blob/master/library.properties#L9 from:
```
architectures=sam
```
to:
```
architectures=sam,samd
```
Which does solve the always updatable issue. If that library will work for either SAM or SAMD then why is a different version included in the SAMD core?

andrei-ivanov, 2017-04-03:
very annoying :-(

PaulStoffregen, 2017-04-03:
We get this problem with Teensy too... the library manager shows a popup that libraries can be updated.  It really ought to be smarter, to only commandeer the user's attention when doing so is truly important.

Especially for beginners having trouble with their projects, sending them down a fruitless troubleshooting path can be a huge distraction from figuring out why their project doesn't work.

TheTechnoGuy, 2017-05-21:
Experiencing the same issue.  Have wasted lots of time on it.

I note that the blurb for the USBHost library indicates that it's intended for use with an **Arduino Due** board.  Not something that I even have... (I have MKR1000 & Teensy boards).

@PaulStoffregen makes really relevant comments about _fruitless troubleshooting path_ & _huge distraction_.  It would be nice to see the Arduino developers take this to heart.
[image]

tabchas, 2017-06-08:
Also experiencing same problem :(

SamuelWAnderson45, 2017-10-04:
Same problem. Arduino IDE 1.8.4

gratefulfrog, 2017-11-13:
same issue, very annoying...

monolied, 2017-11-28:
Same here. IDE 1.8.5, libraries shown but not updatable are "SD by Arduino, SparkFun Version 1.2.0" and "USBHost by Arduino Version 1.5.0". Probably Arduino SAMD Boards conflicting with the Adafruit SAMD boards, which I retrieved using the following Boards Manager URL: https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
This should be easy to reproduce.

andrei-ivanov, 2017-11-28:
I don't think the project is maintained anymore, or just barely.

per1234, 2021-05-15:
Closing as fixed by https://github.com/arduino/Arduino/commit/ac6d3c1afffd81d429498b9d00bc2c92f849cbfe

Prior to that commit, Arduino Library Manager used the installed versions of the [platform bundled libraries](https://arduino.github.io/arduino-cli/latest/platform-specification/#platform-bundled-libraries) as a factor in determining whether an update was available. In addition to not being helpful, since platform bundled libraries are not distributed by Library Manager, but instead by Boards Manager, it also caused this peculiar "always updatable" bug under certain conditions. The change to making Library Manager ignore platform bundled libraries also eliminated this bug.

closed 2021-05-15 by per1234