arduino/Arduino #5284: What is the Process for getting an Issue Resolved

Enigma-TS, 2016-08-23:
Could someone direct me to a webpage/wiki that describes how an issue on the 'arduino.cc' website is reported, evaluated, and any action agreed/decided is then applied so that the issue is resolved?

Last time, I was told you just post it here, but after weeks nothing seemed to have happened (Maybe I was too impatient).

_
As I have identified more issues, I would like to try again.

If the state of the website is considered Low/Ultra-Low Priority (Compared to other activities), just tell me that, and I won't bother anyone.

dlabun, 2016-08-23:
What issue are you referring to?

matthijskooijman, 2016-08-23:
I am not aware of a formally documented process describing what you ask (there might be one internally at Arduino, though I doubt it). In practice, issues get picked up and fixed by whomever has time. This issue tracker is mostly used for issues in the code, issues in the website are also processed, but usually by other people, which I believe might not monitor this issue tracker as closely, so it might take a while (often they stack up and then a bunch of them get processed at a time).

As for you issue, which I presume is #5214, it seems you closed it a week ago, after 9 days of being open, so in this case I think you've been impatient :-). Also, your feedback is slightly complex, e.g. it needs someone to invest a bit of time looking at your files to see what's wrong, which (I would expect) makes it more likely to end up on someone's TODO pile than being fixed right away (as with a simple typo).

Feel free to re-open the issue if it is still relevant. I expect it will be handled sooner or later.

PaulStoffregen, 2016-08-23:
While not documented anywhere, generally it's understood you must post here with enough detail to allow others to reproduce the issue.  Usually this means providing a complete program which can be copied into Arduino and verified or uploaded to a board.  If your post is missing info, probably nobody will respond and the issue will very likely not ever be solved.

If you do post good, complete info, time to solve the issue can vary from a few weeks to many months.  Some issues sit for over a year or more.  Best to be patient, but ask every month or two if more info or detail is needed.

Enigma-TS, 2016-08-24:
Both the already well documented issue (A trace of the Faulty Code's Debug output, plus a Trace of the Proposed replacement code), and a fully commented explanation inside the old code, that easily proves the current code is embarrassingly incorrect.

That seems like a Slam-Dunk, but went nowhere in the 9 days (Sorry, my "weeks" claim was from my obviously poor memory).  I didn't know whether it had got on someone's list or not.

The issue that caused me to Open this thread was a similarly incorrect piece of code (Including a line that does _not_ comply with Recommended Practice in the Reference).
-     "attachInterrupt(1, "  compared to...
-     "attachInterrupt(digitalPinToInterrupt(interruptPin), "
(This approach is designed to be Board-Independent, and a Tutorial has _no_ idea what Board the Tutorial code will be run on)

As a Newbie (New eyes), I have noticed quite a few of these that just aren't right.  My personal reaction is that there are too many (Only my value judgment) issues (Either Coding faults, poor standards, failure to comply with standards etcetera).

My perspective is that if things are in Tutorials, to help Newbies like me, then they should at least me correct.  But, I'm sorry that I have to say, that from what I have seen, and the timeline painted by 'matthijskooijman', correctness doesn't seem like something that is High on the Priority list.

Yes, I realise this is a Volunteer organisation, but 'Volunteer' doesn't have to equate to Faulty and not according to Recommended Practices.  If we only aspire to be sub-standard, we miss an opportunity to 'Go for the Gold' (Sorry, the 2016 Olympics have just finished).

Nobody likes the latest member telling them that there are issues, so I will stop bothering people.

This thread can now be closed in the appropriate manner, by whomever is appropriate.

dlabun, 2016-08-24:
As for #5214, I read your comments and all they do is scream "My code is better than yours". That  tutorial was written by a college professor some time prior to 2011 and there doesn't appear to even be a valid link to it on the Arduino website. This is not commercial grade software to run multi-million dollar LED dimmer systems, it was a basic tutorial to show one way to do something cool and was provide to you at no cost. If you have a better version, contribute to the community and do a Pull Request.

The reason there are so many open issues... A) There's a dedicated group of people who report every valid issue they find and then work like crazy to fix it for the benefit of everyone while not get paid for it and B) There's people who open completely useless issue tickets and then complain nobody is paying attention to them. The core group of developers don't have time to respond to and close pointless issues tickets, the person that opens an issue ticket can close it themselves at any time.

Enigma-TS, 2016-08-24:
Closed as recommended.

closed 2016-08-24 by Enigma-TS

lmihalkovic, 2016-08-24:
@dlabun the core team is comprised of salaried employees