arduino/Arduino #5285: Watchdog problem 8Mhz bootloader

PRosenb, 2016-08-24:
I burned the 8Mhz bootloader to the ATmega328 (on a breadboard) with this sketch:
https://github.com/nickgammon/arduino_sketches/tree/master/Atmega_Board_Programmer

With that bootloader I experience problems with the watchdog reset. Instead of doing a reset, it gets stuck with a fast flashing LED (output 13).

I read about a bug in the standard Arduino Uno bootloader that caused such problem in the past.

May this bug not be fixed in the following bootloader?
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/bootloaders/atmega/ATmegaBOOT_168_atmega328_pro_8MHz.hex

matthijskooijman, 2016-08-24:
I think this is essentially #4492, and indeed the bootloader you reference probably has this bug. I'm not sure what the update policy for bootloader is. The best way forward is probably to switch to optiboot, though that changes the bootloader size in a hard to detect way. @cmaglie, any comments?

PRosenb, 2016-12-20:
@matthijskooijman Is there any progress on the bootloader bug? It still blocks me from using it with 8MHz reliably.
Could you build the Arduino bootloader with the bug fixed or can I download it somewhere?
Thanks for some help.

matthijskooijman, 2016-12-20:
I'm not really the one to ask, so I'll try highlighting @cmaglie and @facchinm again.

facchinm, 2016-12-20:
Switching to optiboot (mainline) for new projects is the way to go. The policy is to release an updated bootloader as soon as it gets flashed on any production board (or if a board has a native way to update it - like the Zero).
Older bootloaders are there for reference/research and are probably FULL of bugs :smile:

per1234, 2017-07-05:
Closing as duplicate of https://github.com/arduino/Arduino/issues/4492.

closed 2017-07-05 by per1234