arduino/Arduino #5286: IDE 1.6.10 - library power.h

pippo-75, 2016-08-24:
Hi all, I report a bug(?) in power.h library, distributed with IDE 1.6.10 Win.
The error is:
exit status 1
'Power_usart3_disable' was not declared in this scope
[image]

The library lacks the definition for USART3. Solved by adding the following lines (from 852 to 855):
#if defined (__ AVR_HAVE_PRR1_PRUSART3)
#define power_usart3_enable () (PRR1 & = (uint8_t) ~ (1 << PRUSART3))
#define power_usart3_disable () (PRR1 | = (uint8_t) (1 << PRUSART3))
#endif

A problem with the -Wire library- (Note: the IDE 1.6.9 compiled all, without errors):
to make sure that the IDE 1.6.10 correctly loads the library > LiquidCrystal_I2C <,
I had to move the content of -src folder- to the root (Wire folder). 

Now compile fine.

facchinm, 2016-08-25:
Hi @pippo-75,
the problem here is avr-libc. This patch (https://github.com/vancegroup-mirrors/avr-libc/commit/12d7b0a7e57ac04d2984b5bfdefb31c21e1389b8) reworked `power.h` but  lost `power_usart3*` functions. Thanks for reporting, I'm escalating this to avr-libc developers.
About `LiquidCrystal_I2C`, I had no problem with the version included in the Library Manager (from @marcoschwartz). Which one are you using?

facchinm, 2016-08-25:
Problem reported and patch submitted here http://savannah.nongnu.org/bugs/index.php?48898

J-M-L, 2016-11-06:
How can this be included in the next build of the Arduino IDE?

facchinm, 2016-11-07:
The patch was applied to avr-libc trunk, but not released yet. We'll ship it as soon as Microchip/Atmel releases a toolchain with the patch applied (or sooner, if avr-libc should publish a bugfix version)

tomjennings, 2017-07-16:
power_usart3_disable() and power_usart3_enable() remain undefined.