arduino/Arduino #5287: HTTPClient POST larger files doesn't work anymore.

comino, 2016-08-24:
### Basic Infos

ESP8266HTTPClient: TCP socket hangs up at sending larger files with. 
With stable version is was working just fine even with very large files so it was introduced lately. 
 
#### Hardware
Hardware:			ESP-WROOM2
Core Version:      	latest github

### Description
Im using HTTPClient.sendRequest("POST", file). 
With smaller files <1kB it works just fine. With larger files ~50kB it failes with HTTPC_ERROR_SEND_PAYLOAD_FAILED after a long timeout.

The error occurs in this part of the code: 

L492-499 ESP8266HTTPClient.cpp
`      
bytesWrite = _tcp->write((const uint8_t *) (buff + bytesWrite), leftBytes);
bytesWritten += bytesWrite;   
           
if(bytesWrite != leftBytes) {
    DEBUG_HTTPCLIENT("[HTTP-Client][sendRequest] short write, asked for %d but got %d failed.\n", leftBytes, bytesWrite);
    free(buff);
    return returnError(HTTPC_ERROR_SEND_PAYLOAD_FAILED);
}
`
Output is: 
"short write, asked for 1460 but got 0 failed"

So there is no more data send via "_tcp->write()" after ~4kB

Any major change in the WiFiClient which may cause this?

sandeepmistry, 2016-08-24:
Hi @comino,

This Github repo does not host the ESP8226 cores and libraries. I would suggest you open the issue here https://github.com/esp8266/Arduino/issues instead.

closed 2016-08-24 by sandeepmistry

comino, 2016-08-25:
sorry :D wrong repo :D