arduino/Arduino #5291: [DOC/WEB] Deprecate libraries related to retired boards (again)

q2dg, 2016-08-26:
Well, https://github.com/arduino/Arduino/issues/3493 was closed without resolving the problem and I can't reopen it, so...I open this one. In summary: reading the information shown in main product page (https://www.arduino.cc/en/Main/Products), I think main libraries page (https://www.arduino.cc/en/Reference/Libraries) should be updated. Specifically:

    -EEPROM library should be moved to a specific "AVR-only boards" section or similar (until it gets generalized to other architectures as well...maybe some day...see https://github.com/arduino/ArduinoCore-samd/issues/145)
    -TFT library should be removed from page and/or moved to a "Retired products related libraries" section or similar. The same for GSM library. The same for (old) WiFi library. The same for "Due only library". The same for "Esplora only library". The same for "Arduino Robot library"
    -Wifi101 library should be promoted to "Standard libraries" section (as it was old WiFi library) but specifying -as it is now well done- it can be used with Wifi101 shield or MKR1000 board only.
    -Ethernet library description should say "Ethernet compatible shield" (or similar) because Arduino Ethernet shield doesn't exist anymore.
     -"Due, Zero and MKR1000 libraries" section and "Zero and MKR1000 libraries" section should be merged into one called like the latter. I don't know if Curie should be included too
     -""USB libraries" section should be renamed to "Micro, Zero and MKR1000 libraries".  don't know if Curie should be included too
      -The reasoning of which libraries are mentioned in "Contributed libraries" and which not maybe should be rethinked...but that's not urgent, I think.

Come on...it's just a minute.
Nowadays this page is a complete mess
Thanks

P.S: On the other hand (should I open a different issue?) , I assume TFT, GSM, Esplora and Robot libraries will be removed in default installations of future IDE's versions, won't they be?

per1234, 2016-08-26:
The Contributed Libraries section is really a mess:
- Messenger: deprecated
- NewSoftSerial: page moved and library merged into the core
- Simple Message System: deprecated
- sserial2mobile: last release in 2010
- Webduino: page moved
- X10: no longer developed and author recommends using a different library
- Debounce: deprecated
- GLCD: deprecated
- Improved LCD library: deprecated
- LedControl and LedControl: Two entries for the identical library
- Matrix: there's no clear way to download it other than downloading the entire Wiring IDE and extracting the library, which won't even compile due to the `Wiring.h` thing.
- Sprite: there's no clear way to download it other than downloading the entire Wiring IDE and extracting the library, which won't even compile due to the `Wiring.h` thing.
- Tone: page moved
- DateTime: deprecated
- PString: page moved

\>50% of the items on that list have issues. That's without even checking if the libraries even work with the current IDE version, which judging by the age of most of them is doubtful. It's obvious that the web team is not going to keep this list current so it should just be replaced with some text like:
> See **Sketch > Include Library > Manage Libraries...** for a large selection of contributed libraries.

Or there's a much better maintained, though far from perfect or complete, list at http://playground.arduino.cc/Main/LibraryList

q2dg, 2017-07-09:
Please, listen to @perf1234 and me...it's very easy solve this issue...maybe 10-15 min.
It's only about having a little care

per1234, 2020-09-29:
The "Contributed Libraries" section has been replaced with an automatically updated listing of all the libraries in Library Manager.

closed 2020-09-29 by per1234