arduino/Arduino #5292: Preferences screen is 650 pixels high, so people with 800x600 resolution can't click OK

SpenceKonde, 2016-08-26:
The preferences screen on the IDE is now about 650 pixels high by my measurement. Hard as this may be to belive, many people are still using 800x600 resolution - like this fellow here: 

http://forum.arduino.cc/index.php?topic=420603

As noted in that thread, there's a huge amount of blank space at the bottom of the preferences window between the options and the buttons at the bottom - simply getting rid of that empty space would allow the preferences screen to be made small enough to fit on people on low res screens to use.

cousteaulecommandant, 2016-09-02:
I'd say that 800x600 is rather uncommon nowadays, however many netbooks seem to use 1024x600 (the Toshiba NB505 in that thread, as well as my Acer Aspire One).  I remember having a similar issue with Chromium preferences there.

per1234, 2017-07-05:
Closing as duplicate of https://github.com/arduino/Arduino/issues/4753

closed 2017-07-05 by per1234