arduino/Arduino #5296: Relative includes of libaries will not work anymore (1.6.11)

rkost, 2016-08-26:
Got compiling issues when upgrading from 1.6.9 to 1.6.11:
Following (very simple) code:

* mainFile.ino *
```
#include "internals/Servo.h"

Servo servo;

void setup()
{
  servo.attach(6, 0, 160); // Numbers doesn't matter...
}

void loop()
{
  servo.write(20);
}
```

In the directory **anywhere/mainFile/internals/ ** are the three files **Servo.h, Servo.cpp, ServoTimers.h** with some adjustments to the includes and timers (they are copied from the original Arduino 1.6.9 lib (**anyhere/arduino1611/libraries/Servo/src/andSoOn**). See attachments for these files:
[Servo.zip](https://github.com/arduino/Arduino/files/439945/Servo.zip)

This code *compiles completly fine* with ardunino 1.6.9 but not with arduino 1.6.11 using this command:
`arduino169 --board arduino:avr:mega:cpu=atmega2560 --verify ./mainFile/mainFile.ino `
(while arduino169 is a soft link to the arduino executable) but failing when compiling on 1.6.11 (doesn't matter if from command line or from IDE's GUI).

*Output for 1.6.11:*
```
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
/tmp/cc6aaBuN.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_Calibration.cpp.o.2111':
cc6aaBuN.ltrans0.o:(.text.startup+0x5c): undefined reference to `Servo::Servo()'
/tmp/cc6aaBuN.ltrans0.ltrans.o: In function `main':
cc6aaBuN.ltrans0.o:(.text.startup+0x150): undefined reference to `Servo::attach(int, int, int)'
cc6aaBuN.ltrans0.o:(.text.startup+0x15c): undefined reference to `Servo::write(int)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

Any suggests? What am I missing?

rkost, 2016-08-26:
Just note that changing `#include "internals/Servo.h"` to `#include <Servo.h>` is doing fine.

matthijskooijman, 2016-08-26:
Since 1.6.10, only .cpp files in (subdirectories of) the `src/` subdirectory will be compiled, so you'll have to move your `internals` directory into `src`. This should also work with earlier versions. I'll go ahead and close this, assuming my answer solves your issue. Feel free to leave a comment if not, then we can reopen.

closed 2016-08-26 by matthijskooijman

rkost, 2016-08-27:
Okay, that was pretty straightforward. Thanks for helping me out on this. Is there any chance to get information about this kind of changes. Don't want to ask silly questions when information is available (could not find them in release notes).

matthijskooijman, 2016-08-27:
Yeah, this should have been made more clear in the release notes. I already pointed this out, but apparently the release notes were not yet updated. @cmaglie, did you have any plans for this still?