arduino/Arduino #5299: Java is killing my Mega2560 and Uno

Cracka79, 2016-08-27:
Arduino: 1.6.10 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

**I receive this massive error when trying to upload a sketch.**

___REMOVE___/bin/avrdude -C___REMOVE___/etc/avrdude.conf -v -patmega2560 -cwiring -P/dev/cu.usbmodem1411 -b115200 -D -Uflash:w:/var/folders/90/c8bdp_nn4zjcqw23t1h4n43w0000gn/T/buildcb64436f2ff7df34155a90f7d8d9aad3.tmp/Blink.ino.hex:i 
java.io.IOException: Cannot run program "___REMOVE___/bin/avrdude": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:130)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:209)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1187)
	at processing.app.Sketch.exportApplet(Sketch.java:1160)
	at processing.app.Sketch.exportApplet(Sketch.java:1132)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2409)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch

PaulStoffregen, 2016-08-27:
Have you tried 1.6.11 or the latest nightly build, to check if this problem has already been fixed?

Cracka79, 2016-08-28:
thanks for the reply. I ran windows 10 on my mac using virtualbox and it worked.  Something to do with the mac setup I guess. All my boards are working again and thats all that matters.......to me anyways.

facchinm, 2016-08-29:
1.6.10 has been quite unlucky, 1.6.11 reverted avrdude to 6.0.1 so it should be ok. Please update to the latest IDE, thanks|

closed 2016-08-29 by facchinm