arduino/Arduino #5301: IDE 1.6.10, 1.6.11 crash when connecting Leonardo compatible ATMEGA32U4 board

ap98nb26u, 2016-08-27:
Hello,

I use a Leonardo compatible board (Da Vinci 32U sold in Japan).  I put supporting files in the hardware directory.  Arduino IDE 1.6.9 was OK, but from 1.6.10, IDE crashes as soon as it recognizes the connected board.  I attach the crash logs and supporting files.  I uses Windows 7 Pro 64bit SP1.

[1.6.10-hs_err_pid11188.log.txt](https://github.com/arduino/Arduino/files/440628/1.6.10-hs_err_pid11188.log.txt)
[1.6.11-hs_err_pid10396.log.txt](https://github.com/arduino/Arduino/files/440629/1.6.11-hs_err_pid10396.log.txt)
[da-vinci-32u.zip](https://github.com/arduino/Arduino/files/440630/da-vinci-32u.zip)

On OS X El Capitan, 1.6.9 also works fine, 1.6.10 causes panics my MacBook.

Best regards,

facchinm, 2016-08-29:
Hi @ap98nb26u ,
On OSX, 1.6.9 and 1.6.10 share the same code regarding serial enumeration, so the problem must be something else.
On Windows, I "transformed" a Micro into a DaVinci using the provided board files. No crash on Win10 and Win8 (but I noticed you are on Win7, which I'm unable to test right now).
Can you try changing usb port on your PC? On Win8 I had to install different drivers because the ones provided are not signed, maybe you could try replacing them with the signed drivers for Leonardo or Micro.

pholmes2012, 2016-09-23:
I am also having this crash issue with a Leonardo compatible (DFRobot Beetle).

I am running Arduino IDE version 1.6.12 on Windows 7. It crashes as soon as the board is connected. Arduino IDE version 1.5.5 does not suffer this problem.

[hs_err_pid5340.log.txt](https://github.com/arduino/Arduino/files/489364/hs_err_pid5340.log.txt)

facchinm, 2016-10-24:
Hi @pholmes2012 , could you try the latest nightly build of the IDE and confirm that the crash in not happening anymore? Thanks

ap98nb26u, 2016-10-25:
Hello @facchinm,
Sorry about my long ignorance, I held my wedding ceremony last month, so I had little time to hack my board.
I just downloaded and tried "1.6.13 Hourly Build 2016/10/21 05:33". It works nicely!
Thank you so much for your effort to fix this and telling me.
I’m so happy to report this.
BTW, from 1.6.10 to 1.6.12 crash here.  hope this helps some.

Regards,

closed 2016-10-25 by facchinm