arduino/Arduino #5302: I can't download 1.0.5 in the web

BZbyr, 2016-08-28:


matthijskooijman, 2016-08-28:
The actual download url is http://arduino.googlecode.com/files/arduino-1.0.5-r2-windows.zip, it seems that these were still at Google Code, which has been closing down for a while. It seems the files are still available at Google Code (https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/arduino/arduino-1.0.5-r2-windows.zip / https://code.google.com/archive/p/arduino/downloads), but it might be better to just move them to the Arduino servers like the newer downloads?

BZbyr, 2016-08-31:
oh! thanks! I got it!

per1234, 2016-09-01:
@BZbyr @matthijskooijman this issue has not been resolved so it should be reopened. The issue has been reported on the Arduino Forum twice in the last week so apparently these downloads are still in demand.

matthijskooijman, 2016-09-01:
Agreed! @cmaglie, can you perhaps make sure this reaches the right people?

mastrolinux, 2016-09-16:
I am solving this and expect to be online on Monday. Sorry for the issue. In the meantime I also increased the download servers capacity so the download should be a bit faster than usual.

closed 2016-09-19 by mastrolinux