arduino/Arduino #5303: Cad file for arduino 101 documentation is not a cad file at all

Donnerschlag, 2016-08-28:
The "cad file" in the arduino 101 documentation tab is not a cad/eagle file at all. It required Allegro to open and I see that it is a confidential file from Intel so I doubt I have permission to have it so I deleted it. Please replace the current cad file with a proper cad file please.

michaelmargolis, 2017-02-13:
It is getting on six months since this issue was raised without being addressed. If suitable cad files are not  provided, I suggest removing the statement on  the 101 hardware page stating that this is open source hardware.

[image]

sandeepmistry, 2017-02-16:
@SidLeung can you please get someone an Intel to look into this?

SidLeung, 2017-02-16:
@sandeepmistry , this issue was brought up in our weekly meeting.  A Jira ticket has been generated to track this item.

var27rag, 2017-03-10:
Hi @SidLeung, what is the progress on this? are we getting a CAD file or will the link be taken down, it is quite confusing.

SidLeung, 2017-03-10:
Jira ticket 848 was created to track the progress of this issue.  Our PM will provide an update here.

per1234, 2018-06-06:
I see the file from that link:
https://content.arduino.cc/assets/Arduino101-reference.zip
Now contains a .brd and a .sch file, which I take as an indication that the "is not a cad/eagle file at all" component this issue has been resolved. I was able to open the files via KiCad's import EAGLE CAD feature.

Although the cover page of the schematic says it's licensed under Creative Commons Attribution Share-Alike license (ver. 3), the board design file still says it's "NOT TO BE REPRODUCED, USED, OR DISCLOSED IN WHOLE OR IN PART TO ANYONE WITHOUT THE PERMISSION OF INTEL CORP." and "CONFIDENTIAL", which doesn't sound very "open source" to me.

@SidLeung was there ever any progress on the Jira ticket?

per1234, 2018-09-26:
@SidLeung any news for us on this?

closed 2021-08-02 by agdl