arduino/Arduino #5305: SPI does not work on arduino Due

fra77x, 2016-08-28:
Please fix the SPI lib.

PaulStoffregen, 2016-08-28:
When reporting any issue, you really must be specific.  Think "reproducible".

Please try to imagine how incredibly busy the Arduino developers must be.  If you do not give the exact code and details to reproduce whatever problem you're experiencing, they have to waste a lot of time trying to guess.

You can and should do much, much better.

fra77x, 2016-08-28:
Yes sorry. I'm trying to use the SPI lib with my Due and i get no signal from my mcp4822 (SPI controlled DAC ic  i 'm testing the Due SPI lib with).

I have asked everywhere and i got no responses so i concluded that probably the SPI lib is broken on the update. I have read every bit of information i could find and the usage of the Lib seems simple. 
Nonetheless i use the SPI and the mcp4822 with my Uno and i can make it work easily. 

But with the Due as many times i tried, different codes, checking all the pins one by one (they work) 
and trying to make everything right. I failed. So i post it here, i hope one of the developers to be able to test the SPI lib with a Due IDE version 1.6.11 sam boards 1.6.9 and if it works i will sincerely apologize.

If it is all right it can't take more than 5 minutes or less to test it..

Thanks,
John

fra77x, 2016-08-28:
I made it work, but it does not work called from an interrupt.

fra77x, 2016-08-28:
Of course i don't want to annoy the "incredibly busy" developers ... Hahaha

closed 2016-08-28 by fra77x