arduino/Arduino #5307: Arduino IDE Crash

HMStith, 2016-08-28:
When I replace the Arduino IDE Example Files with my own Arduino will not start. Is there any way to get around this?

facchinm, 2016-08-29:
Hi @HMStith , 
could you please expand? We need to know 
1 - what you did exactly
2 - your setup (operative system, IDE version, boards connected)

Thanks

HMStith, 2016-09-12:
I was changing the Arduino Example Files in attempt to be my own but the IDE would not start afterwards.

facchinm, 2016-09-12:
Sorry but I can't understand what you are doing. Which operating system are you using? Which IDE version? Can you send the output of "arduino-debug.exe" (if you are on Windows) ? Otherwise I must close the issue as invalid...

sandeepmistry, 2016-12-12:
Closing this for now due to lack of feedback.

closed 2016-12-12 by sandeepmistry