arduino/Arduino #5310: Command line compiling on OSX launches GUI

neilpa, 2016-08-29:
When running `/Applications/Arduino.app/Contents/MacOS/Arduino --verify path/to/sketch.ino` the splash screen is launched in addition to the build. This creates a lot of friction for automating command line builds and integrating with other tools. Is this expected or are their any plans to fix this? My fear is it's a deeper issue with how the IDE is structured given that command line args aren't parsed until after the splash screen (e.g. that's when the invalid param dialog pops). 

Ideally the toolchain/driver could be packaged as a separate JAR for easier integration with build scripts, etc.

neilpa, 2016-08-29:
Looks like what I really want is `arduino-builder`, although it would be nice to use the simpler wrapper here.

per1234, 2016-08-29:
See https://github.com/arduino/Arduino/issues/1970

neilpa, 2016-08-29:
Thanks. Sorry about the dupe but with 600+ it's kinda hard to find things.

closed 2016-08-29 by neilpa