arduino/Arduino #5311: Error in IDE code example: 'Debounce'

OldSteve2, 2016-08-29:
In the 'Debounce' example provided with the IDE, under "File >Examples >Digital", the value of 'millis()' is stored in a 'long' variable type, which is not large enough:-
```
long lastDebounceTime = 0;  // the last time the output pin was toggled
// then:-
lastDebounceTime = millis();
```
It should be stored in an 'unsigned long'.
As a result of this error, many new users have picked up the same bad habit.
Is it possible to have that example modified, and 'long' replaced by 'unsigned long'?
This would save many new users from potential hard-to-detect problems if their code runs for long periods.

Thanks,
... Steve

dlabun, 2016-08-29:
Which IDE version are you using?

OldSteve2, 2016-08-29:
V1.6.9
I originally noticed the error in the Arduino Tutorials section, here:-
[https://www.arduino.cc/en/Tutorial/Debounce](url)
then was reminded that it's also an example provided with the IDE.

closed 2016-08-30 by agdl

OldSteve2, 2016-08-30:
Thank you very much for fixing this. It's very much appreciated.
(Now I can feel comfortable referring people to that example.)

Keep up the good work,
... Steve.