arduino/Arduino #5312: {runtime.tools.arm-none-eabi-gcc.path}/bin/arm-none-eabi-g++": CreateProcess error=2

Gosakan, 2016-08-29:
I am running an STM32F1023c8T6 using an FTDI converter and using it in Arduino IDE 1.6.1. While running the blink program from the examples of STM32 menu, i get this error

{runtime.tools.arm-none-eabi-gcc.path}/bin/arm-none-eabi-g++": CreateProcess error=2

I am new to this board and have no idea to solve. Any help is much appreciated.
Thanks in advance.

[image]

facchinm, 2016-08-29:
Hi @Gosakan , is 1.6.1 a typo? If you are really using that ancient version you should upgrade to a more recent one. Also, to get the `arm-none` compiler, please download `SAM` or `SAMD` core via Board Manager

Gosakan, 2016-08-29:
Thank you. How do i install the compiler via the boards manager ?. Also i chose 1.6.1 because i read that STM32f1 worked only in 1.61 and 1.6.5. Would updating to the latest versions cause any problems ?

facchinm, 2016-08-29:
You can find it under "Tools/Board" menu, at the top. https://learn.adafruit.com/add-boards-arduino-v164/installing-boards
Last time I checked stm32 core was compatible with post 1.6.5 compiler, maybe you could start with 1.6.5 and then test 1.6.11

closed 2016-08-29 by facchinm

Gosakan, 2016-08-30:
Thank you its works fine!@facchinm