arduino/Arduino #5313: [linux] Start with java8

NicoHood, 2016-08-29:
`arduino` on linux is actually a script that launches arduino via java. If you build arduino yourself there is no bundled java included which means it will use your distris java. If its java7, it will crash.

I am maintaining the [Archlinux AUR PKGBUILD](https://aur.archlinux.org/packages/arduino/) which use a trick to ensure arduino starts with java8:
https://aur.archlinux.org/cgit/aur.git/tree/arduino.sh?h=arduino

Would there be another option to include this into the arduino script? It should be distribution independant, so I do not know how its handled on different OS. Does anyone know how we could solve it? I think that should be possible.

@facchinm @cmaglie you possibly have an idea or tip?

cmaglie, 2016-08-29:
> you possibly have an idea or tip?

Sorry no...

BTW this is something that is really distribution-dependent and it should be solved specifically by each distribution like you did. I don't think that there is another way (except bundling the JVM).

(On a side note: I had a quick look to the package, maybe the dependency to the JDK should be set explicitly to the Oracle JVM instead of `java-environment=8` that can be satisfied also with the OpenJDK JVM?)

NicoHood, 2016-08-29:
Why oracle java only? Someone asked me to make it more general and so I used this instead. AUR shows a few wrong packages (java7) but those are not used/valid in pacman, as AUR doesnt check the major version (8 vs 7).

cmaglie, 2016-08-29:
We've had some GUI-lockup problems with the Board Manager: turned out that those lock-ups happened only using OpenJDK (after an update to libatk), instead Oracle JVM works fine.

https://github.com/arduino/Arduino/pull/4362#issuecomment-168029232

I don't know if the problem has been solved in the meantime, that's why I suggested to use Oracle JDK.

NicoHood, 2016-08-29:
As I can see there is no jvm package for arch or am i wrong? However I never had such a problem yet.

closed 2016-09-17 by NicoHood

NicoHood, 2018-02-22:
Some years later arduino is now an official Arch Linux package. Someone pointed out that it should depend on java >=8. Is it true that arduino also works with java 9 installed?