arduino/Arduino #5315: Multiple cursors

MarcelloTheArcane, 2016-08-30:
Not an issue, just a suggestion... can we have multiple cursors in the IDE for typing?

Chris--A, 2016-08-30:
I'm curious as to what you mean.

Do you want to be able to write to two different locations in a sketch at once?

matthijskooijman, 2016-08-30:
I'm also curious, so please explain more.

Also, I couldn't help but think of this famous and silly NCIS scene while reading this: https://www.youtube.com/watch?v=msX4oAXpvUE

MarcelloTheArcane, 2016-08-30:
yeah. like [this](http://blog.brackets.io/2014/04/15/brackets-0-38-release-multiple-cursors/?lang=en).
ok, its strictly a html editor, but i think it would be pretty cool/useful to have it with Arduino. :)

Chris--A, 2016-09-02:
This feature will need to be implemented in https://github.com/bobbylight/RSyntaxTextArea
The text editor part of the IDE uses this component.

matthijskooijman, 2016-09-02:
Also, I think that this is a pretty advanced and complex feature, with a fairly limited scope. If you really need this for your projects, I suggest you use an external editor and use the "use external editor" preference in Arduino. I'm closing this ticket as wontfix, because we'll probably not want to investigate time into this, and *if* it is implemented, it should go through RSyntaxTextArea as @Chris--A pointed out. Feel free to request your feature there.

closed 2016-09-02 by matthijskooijman

MarcelloTheArcane, 2016-09-02:
ok. thanks for the support :+1: